MiSTer-devel/SNES_MiSTer #189: SMW hacks - "The Princess Rescue" & "The Princess Rescue 2 Luigi's Journey" - not working correctly on MiSTer but works on real hardware

Kikusui1, 2020-01-13:
As per title I have discovered two Super Mario World hacks that work on real hardware via an SD2SNES cartridge but do not work correctly on MiSTer. The two real machines I've tested on are a early Japanese Super Famicom & a USA SNES Jnr.

The two rom hacks in question are "[SMW The Princess Rescue](https://www.smwcentral.net/?p=section&a=details&id=21438)" and "[SMW The Princess Rescue 2 Luigi's Journey](https://www.smwcentral.net/?p=section&a=details&id=18801)".

In both cases the intro plays fine on MiSTer but as soon as the game starts with the "welcome box" with text you'll see a corrupted Mario & Yoshi sprite with incorrect colours. Pushing a button moves onto the map screen where you can select a level. Then you'll see the corrupted Mario sprite looking up. There's no animation when you move left to right and corruption when jumping. Running into an enemy crashes the game as does collecting a power up or any interaction with any other sprite / enemy making the games unplayable on MiSTer. 

On the second "Princess Rescue 2 Luigi's Journey" no text appears in the welcome box. You can select a level as normal but then see the same issues as described above.

To build these two hacks an unheadered "Super Mario World (U) [!].smc" rom is needed with MD5 - CDD3C8C37322978CA8669B34BC89C804 / CRC - B19ED489.

Both hacks work on real hardware. I hope reporting these issues on these two hacks brings the SNES core closer to perfection and helps fixes other issues in the process.

paulb-nl, 2020-01-14:
This works properly on release 0106. It seems to be caused by BSRAM: fill with FF a7c06769ba75faf3c87c9721c5f6d9c53826186c because it also does not work on 0108 with SCPU: fix interrupts 362b3e1943308c1a96f3cb671272eaff588b4a23 reverted.

The strange thing is if I use a save file that was created by Bsnes which is filled with FF then it works properly on release 0108 too. 😕 

[SMW The Princess Rescue English.sav.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/4058880/SMW.The.Princess.Rescue.English.sav.zip)

James-F2, 2020-01-16:
Maybe it's better to fix to 0x55 like the SD2SNES for now?
https://github.com/MiSTer-devel/SNES_MiSTer/blob/64ccfb6391825c44ecf4bd462836559abddb4d08/SNES.sv#L607

paulb-nl, 2020-01-16:
That line is for WRAM. This issue is caused by the save RAM (BSRAM). It should be fixed by uneven bytes clear commit.

James-F2, 2020-01-21:
Can be closed?

Kikusui1, 2020-01-21:
I believe it can be closed, these rom hacks are now working for me. 
Thanks everyone for your help.

James-F2, 2020-01-21:
Use the Close button please.

closed 2020-01-22 by sorgelig