MiSTer-devel/SNES_MiSTer #190: Crash during Final Fantasy III attract mode

sqwirral, 2020-01-14:
Core crashes (permanently) at random time during attract/intro/demo mode of Final Fantasy III (USA) (Rev 1).

Usually occurs within 1-2 minutes, sometimes can take longer. Video either goes blank or shows corruption, sometimes flashing and sometimes static. Music usually stops but on one occasion continued.

Occurs with SNES_20190225 onwards. Could not reproduce on any version prior to that, e.g. SNES_20190204 was tested for over 30 minutes and did not crash.

Occurs on both HDMI and direct_video.

Reproduced by 2 users, however 1 other user could not reproduce it.

2 pictures of crash examples: https://imgur.com/a/dixY0hB

paulb-nl, 2020-01-15:
Left it running for an hour without issues. Release 0108. 32MB RAM.

MP2E, 2020-01-15:
I left it running for an hour as well, no crash here.. using latest commit compiled by me, 128MB RAM

tenshin2002, 2020-01-15:
I couldnt get past the the first looping of the attract intro without graphical corruption on Final Fantasy III (USA) (Rev 1).
Corruption usually starts when panning over the Kefka fortress, and when the mode 7 effect when the mechs are walking int the snow the bottom half is black and the falling snow is black. 
The english translation of the Japanese ROM and ran without issues.
Tried current and two previous realeases, all with same issues. 128 MB SDRAM

James-F2, 2020-01-16:
Running the intro for 5 hours here, no crash.

sqwirral, 2020-01-16:
Solved, crashing was overheating from inadequate cooling. I really apologise for wasting all your time. Have confirmed same issue with tenshin2002 too.

closed 2020-01-16 by sqwirral