MiSTer-devel/SNES_MiSTer #195: Megaman X2 / Rockman X2 attract mode desyncs from input

Kreeblah, 2020-01-19:
Something seems to be up with the CX4 implementation.  The intro on Megaman X2 and Rockman X2 demonstrate this in the attract mode with the boss battle.  X isn't supposed to die.

https://www.youtube.com/watch?v=tXrGIChG9Rs has a capture from an Analogue Nt running an actual cart from before they had special chip support for the jailbreak firmware.  You can see that the actual cart results in the same inputs playing out differently, where it stops just as X fires the last shot to win the fight.

ikari's had trouble with this on the SD2SNES as well, which you can see at https://sd2snes.de/blog/faq#comment-990669.  X also dies on an SD2SNES, but the battle does play out slightly differently.

paulb-nl, 2020-01-19:
It is broken since 20200106 but it works correctly with release 20191117.

Kreeblah, 2020-01-20:
Ah, gotcha.  Yeah, I'm on 20200108.

James-F2, 2020-01-21:
Only CX4 related commit I see is this: https://github.com/MiSTer-devel/SNES_MiSTer/commit/0d79e5858aefb536f32e86bb97748eba159746f7

paulb-nl, 2020-01-21:
Reverting that commit cx4: support for 4MB ROM. had no effect but reverting '65C816: fix branch opcodes timing' bc6943c0786eb4922220ffabe6c9134e38b2399c does fix this.

James-F2, 2020-01-22:
That 65C816 fix, fixed Mecarobot Golf.

srg320, 2020-01-22:
This fix bc6943c is needed for accurate 65C816 timings, it was my huge bug. And it seems to be hiding the issue in CX4. I think the CX4 instructions have a different number of clock cycles, but I did not find information about it. If for example reduce CX4 frequency to 18MHz, then demo works fine.

paulb-nl, 2020-01-22:
Here is more info about CX4 timing. Have you seen it? http://forums.nesdev.com/viewtopic.php?f=12&t=14647

sorgelig, 2020-01-22:
Or we can go a simpler route: only 2 games use CX4 - if they are ok with 18MHz, then problem is solved.

James-F2, 2020-05-30:
Fixed in 20200530.
Can be closed.

closed 2020-05-30 by sorgelig