MiSTer-devel/SNES_MiSTer #196: Wild Guns - Constant Flicker and Slowdown.

SegaSnatcher, 2020-01-20:
This issue was spotted on TheHandleBarGamer's recent stream, but because of his capture setup you don't see the actual flickering.  The game also runs well below normal speed.  

https://www.youtube.com/watch?v=ZkKCiWrW9to&feature=youtu.be&t=2116

I personally tested and confirmed using 20200108 release.

paulb-nl, 2020-01-20:
Use the 20191117 release for now. 20200106 and 0108 have issues.

Peredonov, 2020-01-21:
> Use the 20191117 release for now. 20200106 and 0108 have issues.

What is the best way to revert to an earlier release? I tried manually replacing 20200108 with 20191117 and the core would not load, neither would it after manually replacing 0108 back. Using the update script was the only way to be able to load the core again.

James-F2, 2020-01-21:
Latest commit fixed this.
https://github.com/MiSTer-devel/SNES_MiSTer/commit/f62491cedd5b563084e2d63b0898c9b162324f4c

paulb-nl, 2020-01-21:
> What is the best way to revert to an earlier release?

Just copy the rbf to the folder. I use WinSCP with SFTP.

SegaSnatcher, 2020-01-21:
I tried a test build with latest commits and it did indeed fix this issue.  Thanks

closed 2020-01-21 by SegaSnatcher