MiSTer-devel/SNES_MiSTer #200: First door in Super Castlevania IV doesn't open

Chocobobobo, 2020-01-22:
Problem reproed both with the Japanese and US version.
This issue was also referenced here on Higan (interesting I have the same issue on MiSTer).

https://www.reddit.com/r/castlevania/comments/9bekaa/help_cant_open_doors_in_castlevania_iv/

Door with the issue:
https://imgur.com/a/CjWqu0N

Pressing up should open it, but nothing happens. And yes, I have tried everything 20 times, and up works properly.

Chocobobobo, 2020-01-23:
UPDATE
Don't ask me why, but I found what was going on. I tried to play Naxat Pinball Jaki Crush, and got a freeze after the first press start. After trying to load NBA JAM (Japan), I got an error message mentioning a non supported device. It appears that I had the mouse and SS6 enabled. 

After disabling the mouse and SS6 in the core options, all the problems were gone, including the locked door in Super Castlevania 4. How are those related is a complete mystery to me.

It might be good to disable mouse and SS6 for unsupported games by default, or have the option to auto disable it.

Games are really reacting weirdly with those peripherals enabled in the core options...

James-F2, 2020-01-23:
> It appears that I had the mouse and SS6 enabled.
> 
> How are those related is a complete mystery to me.
>
> It might be good to disable mouse and SS6 for unsupported games games by default
:man_facepalming: 
The developers are busy people for such waste of time.

sorgelig, 2020-01-23:
It might be good that users would organize themselves and don't put their problems to developers.
Mouse and Super Scope are disabled by default, so only you is guilty to enable them permanently.

closed 2020-01-23 by sorgelig