MiSTer-devel/SNES_MiSTer #201: blargg's SPC test ROMs (for reference)

James-F2, 2020-01-24:
https://forums.nesdev.com/viewtopic.php?f=12&t=18005
[blargg_testroms.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/4109309/blargg_testroms.zip)

spc_dsp6.sfc fails on MiSTer on "$F0-$FF are not ram".
This DSP6 test is huge, and also fails on or freezes on the 3Chip console.
Passes on higan/bsnes.

wwark, 2021-06-23:
I put the archive link https://archive.is/Mz9Vc because the one for Kishin Douji Zenki - Tenchi Meidou (Japan) doesn't work anymore.

Max833, 2023-02-01:
Archive link for "Kishin Douji Zenki - Tenchi Meidou (Japan)"
https://archive.is/Mz9Vc

wwark, 2024-09-18:
Kishin Douji Zenki - Tenchi Meidou works now with the commit of 18/09/2024.