MiSTer-devel/SNES_MiSTer #202: Secret of Mana - Can't go left or right after loading game

bootsector, 2020-01-29:
Sometimes after loading a game save in Secret of Mana, I can't go left or right, although the character seems to change directions when going left or right:

https://www.youtube.com/watch?v=7wdlNIV_080

It doesn't happen all the time, but resetting the core and letting the the intro rolling a little bit more sometimes makes it happen. I have attached my save game:

[Secret of Mana (USA).sav.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/4127647/Secret.of.Mana.USA.sav.zip)

When hitting OSD, left and right works as intended, so it doesn't look like a controller issue.

I couldn't test this on real hardware nor I couldn't find this issue online, so I'm not sure if it's a known issue with this game.

bootsector, 2020-02-01:
Just another info: this issue happens with a 128MB RAM module. I couldn't replicate it with a 32MB module. 128MB RAM module passes memtest at 140 for one hour with 0 errors.

bootsector, 2020-02-02:
Ok. With 32MB I'm able to go left and right, but there's some graphical corruption if I move to the left inside the Inn after loading the save. :(

bootsector, 2020-04-03:
So Ive got another 128MB RAM module. With Secret of Mana, my old 128MB module had graphics corruption only after loading a save.

With this new RAM module, it gets graphic corruption even from starting a new game:

https://www.youtube.com/watch?v=xVnuu1ARH1g

Both of my 128MB RAM modules pass memtest at 140Mhz for two hours. This module even works with the FF3 intro (https://github.com/MiSTer-devel/SNES_MiSTer/issues/204).

Nothing of this (Secret of Mana or FF3 intro issues) happen on SNES core release 20190204 (a really old one!).

James-F2, 2020-05-02:
I tried and tried and I can't reproduce.
32MB

bootsector, 2020-05-02:
It rarely happens on 32MB. This is for 128MB only

James-F2, 2020-05-02:
Try the newest MemTest

bootsector, 2020-05-02:
Already tried. As usual, it gets stable with no errors at 140Mhz for a long time.

James-F2, 2020-05-30:
@bootsector 
Please check with latest release, and close issue if fixed.

sorgelig, 2020-05-30:
already confirmed on forum.

closed 2020-05-30 by sorgelig