MiSTer-devel/SNES_MiSTer #204: Final Fantasy III Intro

molnara, 2020-01-31:
The text in the intro is all garbled up.  I've tried many versions, all the same.  Latest mister binary and snes core.

[image]

[image]

[image]

bootsector, 2020-02-01:
I've got graphics corruption (different than yours, but definitely corrupted) with a 128MB module. But everything was fine with a 32MB RAM module. Weird.

sqwirral, 2020-02-02:
Improve your cooling
Also see https://github.com/MiSTer-devel/SNES_MiSTer/issues/190

bootsector, 2020-02-02:
Nothing to do with cooling. For the record, as @sqwirral has suggested, core version SNES_20190124 plays FF3 intro perfectly with my 128MB RAM module.

bootsector, 2020-02-03:
In my case, the latest working version of the core is 20190204.

I've made a video of what's happening, if that helps:

https://youtu.be/1UKeg4RS0Hk

It's a very low quality video, but it should give an idea of what's going on here.

James-F2, 2020-02-03:
So nobody noticed for a whole year?

molnara, 2020-02-03:
> 
> 
> Improve your cooling
> Also see #190
> Secret of Mana can also give issues without a fan. These issues vary from one nano to another.

I have a heat sink and io board with fan.  Not sure what else I can do?

bootsector, 2020-02-03:
@James-F2 yeah, this doesn't happen with all RAM modules, but definitely this affects several users (myself included). Versions older than one year doesn't seem to have this issue for any users though. I wonder what change may have caused this... :thinking:

molnara, 2020-02-03:
99% sure I had no issues with FF 3 with my old 32mb sdram, but I'll pop it back in to verify and try out those cores you mentioned @bootsector .  Both my sdram I got from misteraddons too btw.

bootsector, 2020-02-03:
With my old 32MB I've got graphics corruption in SoM only so far. FF3 intro runs good with it. With 128MB I've got this issues/202 in SoM and the FF3 intro graphic corruption.

bootsector, 2020-02-04:
With the help of the Discord guys, we've drill down to the latest working commit: https://github.com/MiSTer-devel/SNES_MiSTer/commit/b90539d3e31f3c00c19f17f55fa95a4180df87c1

Everything after that commit is either black screen when loading ROM or the issue I'm showing above in the video.

bootsector, 2020-02-04:
For the record, this is the commit that gives me the issue I'm consistently having and shown in the video above:

https://github.com/MiSTer-devel/SNES_MiSTer/commit/53be7b8fdcec648ac344701178a99104b8e032ae

Everything else between the last working build and this commit is either black screen or garbled graphics as soon as I load the ROM.

sorgelig, 2020-02-04:
i don't see any meaningful change in that commit. It's cosmetic change.

mathieulh, 2020-02-04:
As far as I can tell, this;: https://github.com/MiSTer-devel/SNES_MiSTer/commit/dea81ccfc5e323f3f4b8bea816bca3a25fbb6fcb#diff-ad0bf9f31f9747488d62865beb7d9daaL902

is the exact commit that breaks things.

bootsector, 2020-02-07:
@sorgelig if that helps finding and fixing where the issue lies, I can send you my DE10 + ram module. let me know!

molnara, 2020-02-08:
I got a chance to swap out my 128mb sdram with my old 32mb sdram and now the intro looks fine.  I'll try the other older cores as suggested with the 128 and will report back.

gyurco, 2020-02-12:
This one:
https://github.com/MiSTer-devel/SNES_MiSTer/blob/master/sdram.sv#L53
looks bad. Might work on some SDRAM chips, might not on others.

sorgelig, 2020-02-12:
Because of tight timings i had ti reduce that time. For officially recommended Alliance SDRAM P/N it works fine.
Links to 1-year old commits which caused the issue shows it's not related to RASCAS delay.

bootsector, 2020-02-12:
Is there a way to change that to a safe value so I can test? Since we're all clueless, anything would help on this investigation.

sorgelig, 2020-02-12:
> Is there a way to change that to a safe value so I can test? Since we're all clueless, anything would help on this investigation.

if you have time, then you can travel back through releases first to find the rough place where your 128MB module started to fail. Then between working and non-working release go through commits to find the commit breaking your SDRAM module. There is possible some commits may breake some other things than what you explore, so don't care of other problems.

bootsector, 2020-02-12:
This is what I could find so far when trying intermediate commits:
`
https://github.com/MiSTer-devel/SNES_MiSTer/commit/b90539d3e31f3c00c19f17f55fa95a4180df87c1 - Reduce audio taps.-> Works

https://github.com/MiSTer-devel/SNES_MiSTer/commit/c4c607b5aa1947e7c5da245566a873c31201fb86 - Fix Vsync length -> Blackscreens as soon as I load the ROM

https://github.com/MiSTer-devel/SNES_MiSTer/commit/fee8ad04d2e6c37371ead0dc2a03043b79623345 - Merge pull request #35 from paulb-nl/vsync -> Blackscreens as soon as I load the ROM

https://github.com/MiSTer-devel/SNES_MiSTer/commit/20e86d40c9d097c476aa56d62872f92b09a8d414 - Merge pull request #14 from MiSTer-devel/master -> Blackscreens as soon as I load the ROM

https://github.com/MiSTer-devel/SNES_MiSTer/commit/deffe4359f6c899b0e2380027f31863fc302e19b - SA1 support -> Gets garbled graphics as soon as I load the ROM 

https://github.com/MiSTer-devel/SNES_MiSTer/commit/7decade75e189310936362ae37e99d4a8bf31902 - Merge pull request #36 from srg320/master -> Blackscreens as soon as I load the ROM

https://github.com/MiSTer-devel/SNES_MiSTer/commit/dea81ccfc5e323f3f4b8bea816bca3a25fbb6fcb - sys: update scaler, scandoubler. User port definition. Remove VIP. -> Doesn't synthetize on Quartus

https://github.com/MiSTer-devel/SNES_MiSTer/commit/53be7b8fdcec648ac344701178a99104b8e032ae - Minor update. -> Fails just the same as latest release, same issue at the same spot.

https://github.com/MiSTer-devel/SNES_MiSTer/commit/9bb724e995263b892f9e987d7153c3218ed34642 - Rewrite the muxer of chips. -> Fails just the same as latest release, same issue at the same spot.

All subsequent commits -> Fails just the same as latest release, same issue at the same spot.
`

paulb-nl, 2020-02-12:
If it breaks by just changing the Vsync length then it must be random timing issues and you should get the same results by changing the fitter seed between builds.

Assignments->Settings->Compiler Settings->Advanced Settings(Fitter)->Fitter Initial Placement Seed

sorgelig, 2020-02-12:
Common, these commits are way too old! SDRAM v2 (128MB is v2) wasn't supported at that time!

tenshin2002, 2020-02-12:
I have same kind of issue on the Final Fantasy III (USA) (Rev 1) rom. In the intro i get corrupted graphics in the second or third loop of the intro demo. The strange thing is that Final Fantasy VI - FF3 Retranslation (J) T+Eng 1.2b RPGONE rom runs without issues

tenshin2002, 2020-02-13:
I just tried this build that Mathieuhl built for testing purposes and it didnt fix my issue:
https://github.com/mathieulh/SNES_LLAPI_MiSTer/commit/c1681dca207ac97e2a1b3ed95fcdef8cc725180a

kupad, 2020-03-26:
I can confirm that I see a graphical glitch after the first loop of the intro as well.

I have a heatsink and fan, and 128 SDRAM.

joshuacant, 2020-05-18:
Is it possible this issue could be related to the vsync_adjust value in mister.ini? The issue was consistently there when it was set to 2 or 1, and then resolved after setting it to 0.

You may want to check your ini file and try out 0 if that's not your current setting.

bootsector, 2020-05-18:
@joshuacant changing that didn't fix it for me.

molnara, 2020-05-18:
@joshuacant I've tried every setting.  I've come to the conclusion it is indeed my 128mb sdram and I doubt this will ever get resolved.  I'm back using my 32mb sdram with no issues.

molnara, 2020-05-27:
Reopening because of renewed [interest](https://misterfpga.org/viewtopic.php?f=18&t=90)

molnara, 2020-05-29:
Latest [build](https://misterfpga.org/download/file.php?id=74) by srg320 fixes the issue for me.  Please test and confirm.

bootsector, 2020-05-29:
It fixed for me as well

sorgelig, 2020-05-30:
fixed in new release.

closed 2020-05-30 by sorgelig