MiSTer-devel/SNES_MiSTer #207: WILD GUNS: Unplayable after starting game, heavy flickering and slowdown

cpmattes, 2020-02-09:
Like mentioned in title, after starting US or JAP Version of Wild Guns with SNES_20200108.rbf everything seems to be fine, title screen and menu are flawless, controller is reacting. After choosing a character and starting the game, the picture is flickering heavily and gameplay is evenly slowed down. Using 128MB ram expansion, setup cooled passive and with a noctua.

paulb-nl, 2020-02-09:
This has already been fixed.

@sorgelig please make a new release

sorgelig, 2020-02-10:
i didn't release yet because i see debates when one commit fixes one thing but breaks others. So i'm unsure if it's ready to release or not.

paulb-nl, 2020-02-10:
It is ready for release. The regressions have been fixed.

sorgelig, 2020-02-11:
alright. will release tomorrow

Left-Empty, 2020-03-12:
Hello,
I have just tried it, and the issue seems to still be here.

paulb-nl, 2020-03-12:
Works fine here:
https://youtu.be/UZvSomwdStI

cpmattes, 2020-03-12:
Last update fixed it for me!

Left-Empty, 2020-03-12:
It seems my wonderful little MiSTer wants me to look bad here: for two days the game exhibited the glitch. 
I try again today after watching paulb-nl's video, using the US version: it works. I tried again my Japanese version. It works.
I apologize.

SegaSnatcher, 2020-04-13:
> Last update fixed it for me!

Probably best to close this issue now.  Thanks

closed 2020-04-14 by sorgelig