MiSTer-devel/SNES_MiSTer #208: Unknown sync issue when compared to real console

Ownasaurus, 2020-02-12:
Hello,

This is Ownasaurus from the TASBot community. We make TAS replay devices to test Tool Assisted Speedruns on console. I am currently testing my replay device, TAStm32, with the MiSTer cores to test for accuracy (using a SNAC).
NES syncs great. However, SNES does not appear to sync. I am using Super Mario World (USA) to test. As-is, it does not sync. I am using http://tasvideos.org/3019M.html as my test case. If I prepend one blank frame, it makes it all the way to the water level. This means it may be eating one extra frame on startup? Not entirely sure. Once I reach the water level, many frame adjustments need. I suspect there may be an issue in the core.  I am not sure if it is an issue with polling or with the emulation of loading times or something. How might we be able to look into this further?

paulb-nl, 2020-02-12:
Have you tried some older versions of the SNES core?

Ownasaurus, 2020-02-12:
No, I have not. Are there particular versions you would recommend I try?

paulb-nl, 2020-02-12:
Try from 20191117. There is also a new release today.

Ownasaurus, 2020-02-13:
I was using today's release (compiled from source with minor mod to ensures serial  input is taken as player1). will try 20191117 next

Ownasaurus, 2020-02-13:
The old version (20191117) appears to be significantly worse. The newer version can make it to the 3rd level of SMW in this run: https://www.youtube.com/watch?v=hboxJ_z-P7E . the older version can't make it into the first level.

paulb-nl, 2020-02-13:
I guess frame by frame comparisons are needed. Could you capture a video of the desync?

Ownasaurus, 2020-02-13:
Unfortunately I can only capture video at 30fps I think. I can share a logic analyzer capture though which shows a difference in polling.

Ownasaurus, 2020-02-13:
Here are some measurements from my LA dump that may help you:
MiSTER core (latest):
Latch pulse duration: 4.9us
Time from one rising latch edge to another rising latch edge: 16.64ms
Clock pulse duration: 6.91us
Time from one falling clock edge to another falling clock edge: 11.92us

Real SNES (US):
Latch pulse duration: 12.04us
Time from one rising latch edge to another rising latch edge: 16.64ms
Clock pulse duration: 5.9us
Time from one falling clock edge to another falling clock edge: 11.91us

Conclusions: your latch pulse should probably be wider and your clock pulse should probably be narrower. However, period of the latch and clock pulses seems to be spot on.

I hope this helps. This is likely not even the reason for the desync... but it is some info to make the core more accurate!

Ownasaurus, 2020-02-13:
[image]
Here is a screenshot to illustrate another difference. Note that the MiSTer using the SNAC (bottom part of screenshot) has an extra latch compared to real console at the right part of the screenshot. This part of the LA dump showed the first few latches from poweron. I suspect there will be other discrepancies later on too.

I hope this helps.

paulb-nl, 2020-02-14:
Thanks!  Can you try a build with the ACLK_FREQ set to 410112? 
https://github.com/MiSTer-devel/SNES_MiSTer/blob/34c0a743cee9e1277c37f6fc682c39be0b7dad0c/src/DSP_PKG.vhd#L10

With the current value I see that the auto joypad polling is kept enabled for 2 frames where that extra latch is in your logic analyzer capture.  It is enabled at line 232 and disabled at line 239 the next frame. It seems to be related to the SPC700 timer.

By matching the SPC700 clock speed to Bsnes, joypad polling is enabled at line 213 and disabled at line 239 the same frame. That is much closer to Bsnes.

Ownasaurus, 2020-02-15:
Thank you, Paul, for this suggestion. However this fails at the same place. Additionally, I have to prepend two blank frames for the beginning to line up correctly. (In the previous version I had to prepend one blank frame.)

Ownasaurus, 2020-02-15:
On a real SNES, I get the correct behavior for this run if i prepend any odd number of blank frames (including 0). on mister, the behavior seems to vary.
Ilari (lsnes dev) says that bsnes uses "32040*768" for the ACLK freq. does this number mean anything to you? I'm not very familiar with snes internals unfortunately.

Kitrinx, 2020-02-15:
It seems that on real console, a second crystal of 24.576mhz is used for DSP which then clocks the SMP. On mister it is using the 21 mhz master clock with a clock divider for the SMP's CE. I'm not very familiar with SNES, is using the wrong clock for DSP accounted for in all places, or only the CE? Shouldn't the CE then be 4.096mhz?

Kitrinx, 2020-02-15:
Because the SPC clock seems to be divided by two in the polynomial divider, is it possible it is out of phase by one clock (it should be ~8mhz)?

paulb-nl, 2020-02-15:
The 410112 freq that I suggested is 32040*128 so it matches Bsnes. The extra latch should be gone with that. With that change joypad polling at that last part is enabled at lines 213-239 vs 214-239 in Bsnes.

If changing the audio clock does not affect the water level then maybe there are other timing issues.

Can you capture some videos? Even at 30fps it might reveal some differences.

sorgelig, 2020-02-15:
ACLK_FREQ was empirically tuned by Bad Apple demo. With 409600 the audio disappear at some point of demo.
Otherwise you can set 409600 and see if problem is fixed or not.

Ownasaurus, 2020-02-15:
Hi paul are you available in discord to talk in realtime? this way i can capture exactly what you need. at this point i have ~12 builds, so i'd like to know what you'd ideally like captured

Ownasaurus, 2020-02-15:
Thanks for that info, sorgelig. Do you know if the Bad Apple demo runs correctly on bsnes or console by any chance?

paulb-nl, 2020-02-16:
I am not on Discord. Just a capture with the  standard 0212 release from the beginning of the run until it desyncs in the water level is fine. Unless the SPC clock change makes it further in the run.

There are at least 2 Bad Apple demos. On one the audio stops in the middle of the demo depending on the console.
https://github.com/MiSTer-devel/SNES_MiSTer/issues/146

The other was updated to maybe work on real hardware. https://forums.nesdev.com/viewtopic.php?p=243111#p243111

sorgelig, 2020-02-16:
> Do you know if the Bad Apple demo runs correctly on bsnes or console by any chance?

I don't know. I have no real SNES.

> The other was updated to maybe work on real hardware. https://forums.nesdev.com/viewtopic.php?p=243111#p243111

Thanks for info! I will test on this version.

Ownasaurus, 2020-02-16:
https://youtu.be/uGs86263sgI standard latest release (with trivial modification to ensure accurate playback)

Ownasaurus, 2020-02-16:
https://www.youtube.com/watch?v=hboxJ_z-P7E here is a reference rendering. this is how it plays on my real SNES console too without any blank frames prepended

sorgelig, 2020-02-16:
> The other was updated to maybe work on real hardware. https://forums.nesdev.com/viewtopic.php?p=243111#p243111

This version works fine with 409600. This version looks like 30fps (instead of 15fps on other version) but audio is like portable radio from 80x. But since even original HW stops the audio on 15fps version, then it's better to use 409600 to match the original.

Kitrinx, 2020-02-16:
Having spoken to Byuu about this, it seems the 24mhz ceramic oscillator on the original system was notoriously inconsistent. He claims (in bsnes) using 409600 will break two games, but it seems maybe that is a symptom of bsnes more than anything else. If it was indeed inconsistent on a real system, then 409600 should be fine on our recreation as well.

Ownasaurus, 2020-02-16:
[image]
Top = real console. Bottom = MiSTer at 409600. I manually searched for the first discrepancy. Note that MiSTer lags an extra frame at the indicated portion. Everything seems exact until then.

Ownasaurus, 2020-02-16:
Of note, a small change to 409610 shows the same "first error" depicted above.
However 409600's first visibly noticeable desync is in the 2nd level.
Whereas 409610's first visibly noticeable desync is before it even enters the 1st level (fails overworld movement).

paulb-nl, 2020-02-16:
How does your TAS device sync to the SNES? Does it use Vsync, Latch pulse or something else?

Ownasaurus, 2020-02-16:
Latch pulse.

Ownasaurus, 2020-02-16:
To clarify, my device acts like a real controller and only interfaces through the controller port. When it receives a latch, it shifts out bits on the clock pulses. The only difference between my device and a real controller is that the input is pre-determined from a file rather than given by a human.

Ownasaurus, 2020-02-16:
[image]
Top is real console. Bottom is 410112. First difference in polling is noted around 8._something seconds in each dump. they are synced until then... lag frames and all. Therefore 410112 says in sync longer on SMW

paulb-nl, 2020-02-18:
How consistent are these Mario world TAS runs between consoles? If the clock phase between CPU and SPC700 is different between consoles then it should also desync on some consoles. 

On a Super Metroid console verification he was saying it was very difficult to sync and maybe it wouldn't work because he had a different console.

Ownasaurus, 2020-02-18:
Yeah most of the games are very inconsistent. However, SMW works fantastic on every console I've tested. That's why I'm using it as the test case. It seems to not be as sensitive for some reason.

Ownasaurus, 2020-02-18:
Which, by the way, is also why i am shocked that a change of "10" to the ACLK affected the results. It makes me suspect there may be another issue.

paulb-nl, 2020-02-18:
I noticed that Bsnes latches the auto joypad polling register at the beginning of Vblank so polling should not start when it is enabled in the middle of Vblank.

Can you try a build with the commit below? It has a latch added at the beginning of Vblank that should prevent those extra latches.
https://github.com/paulb-nl/SNES_MiSTer/commit/c9684828762e9550dbdaf1e3f4e6d184e05d0be8

Ownasaurus, 2020-02-19:
What a significant improvement that made! With the polling change as suggested and an ACLK of 410112, mario makes it several levels further. The desync ends up occurring in the ghost house where mario tries to clip through the diagonal staircase at the beginning of the level.

However, careful analysis of the LA dump shows that MiSTer has an extra lag frame at the end of the first level compared to console: 
[image]

That is the only difference I see. It holds that until it eventually visibly desyncs in the ghost house.
Here is a video recording of the run: https://www.youtube.com/watch?v=e-4ndQdzSfc

paulb-nl, 2020-02-19:
Great that it makes it further but not so great that it still desyncs :smile:

I don't think that lag frame after the first level is a problem if it stays synced all the way to the ghost house?

The inputs at the ghost house seem to happen 1 frame too early so I would have expected a latch difference near there. Interestingly the subtitle there says "1 lag frame saved by luck in the transition".

Ownasaurus, 2020-02-19:
It is very possible I just missed a difference in my manual analysis.
Tried a build with A=412220 (slightly higher), and it made it half way through star world. I can get a capture if you'd like. Or I can try and further tune it.

paulb-nl, 2020-02-19:
412220 is about 32.204Khz. I don't know if the oscillator could be so far off. 

Since you said the TAS works on all SNES consoles that you tried it should not be so sensitive to the SPC clock speed.

Ownasaurus, 2020-02-19:
Yeah it IS rather odd... i do not know how to explain these findings since I do not know the SNES internals too well. i can only collect and report the data :( Unless perhaps there were another bug somewhere in there? With the clk dividers perhaps?

Ownasaurus, 2020-02-19:
Oh i'm sorry Paul I had a typo!!!  I meant 410220. Only a slight difference from 410112.

Ownasaurus, 2020-02-27:
Any suggestions on where to go from here? Here are some ideas I have:
- I could open up my snes and measure its xtal
- I could use my LA to measure the input polling relative to the analog video signal to confirm that the real snes polls at the beginning of vblank
- I could modify bsnes's ACLK freq to see if the SMW run is equally as sensitive. If not, that suggests another underlying issue

ghogan42, 2020-03-01:
In case it helps, here are some recently measured clock rates from SNES hardware.  I think they are still a bit higher (avg 32066) compared to what has been tried here?

These are from logic analyzers and done my Artemio (author of 240p test suite and MDFourier) and plgDavid (reverse enigineered many synth chips).

Owner/tester | S-APU type | Serial/info | Clockrate | Cold Boot
-- | -- | -- | -- | --
  |   |   |   |  
Artemio | SHVC-SOUND (no rev) | S11345226 | 32080.3 | 32076.8
Artemio | SHVC-SOUND revD | UN15113630 | 32075.9 | 32070
Artemio | SHVC-SOUND revF | UN15605752 | 32075.4 | 32073.9
Artemio |   | UN316642987 | 32073.1 | 32069.7
Artemio |   | UN314757546 | 32087.2 | 32083
Artemio | JR | UC308298420 | 32028.9 | 32020.3
plgDavid | SHVC-SOUND revA | (loose on bench lost track) | 32097 | 32089.2
plgDavid | SHVC-SOUND revC | chipsynth SFC test reference | 32108.4 | 32100.7
plgDavid | SHVC-SOUND revD | (loose on bench lost track) | 32018.3 | 32010.7
plgDavid | SHVC-SOUND revE | (loose on bench lost track) | 32111.3 | 32104.9
plgDavid | GMP-01 | UN20747821 | 32017.7 |  
plgDavid | JR | UC600011420 | 32040.3 | 32032.4
infiniteneslives | SHVC-SOUND revC |   | 32038 |  
infiniteneslives | GMP-01 |   | 32035 |  
Lugerius | GMP-02 | UN234813674 | 32119 | 32116.9
Lugerius | JR | UC204851415 | 32058.9 | 32051.9
  |   |   |   |  
  |   | Average | 32066.54375 |

Ownasaurus, 2020-03-02:
OK. So they seem to vary maybe +- 10 but not hundreds.
How do these numbers relate to the ACLK value I have been fiddling with, such as 410112?

paulb-nl, 2020-03-02:
32040 x 128 ÷ 10 = 410112

> I could modify bsnes's ACLK freq to see if the SMW run is equally as sensitive as it is in the MiSTer core. If not, that suggests another underlying issue

It would be interesting to see that result.

Ownasaurus, 2020-03-02:
Paul,

According to Ilari who developed lsnes (uses bsnes core):

> just changing audio clock makes movies break even if they would work with that frequency on real console played by bot. The emulator syncs to video frames, the bot syncs to input frames. And the two are not the same when loading with autopoller off is involved.

So according to him, even if changing the clk in bsnes makes a move desync, it will not necessarily desync on real console. Sounds like MiSTer SNES core is planning on handling input more like a real console. Will look into this more as time allows.

Ownasaurus, 2020-03-03:
In bsnes core using lsnes, the clk is set at 24607104. If I change the clk to, 24608104 (1k frequency difference), it desyncs in the second level of the smw-bruno run. This confirms that the emulator is indeed extremely sensitive -- more sensitive than the actual clk variance on console.