MiSTer-devel/SNES_MiSTer #210: Megaman X Sound Issue 20200212

WonkerChonker, 2020-03-02:
Howdy

I notice a strange buzzing/hard-to-describe-vibrating noise when booting up this game and listening to the music. It is very noticeable when playing the intro stage (01 in sound test I think). Booted up the analogue super nt and the audio is clear and correct. Both systems by HDMI directly into the same port on a reciever. No audio artifacts noticed in SMB all-stars. Mister is being cooled with heatsink and IO board fan.  Maybe it's just me or my setup but I figure if someone else notices it then it probably aught to be brought to the attention of the SNES core dev.

sorgelig, 2020-03-02:
Try to load this game first after loading SNES.

paulb-nl, 2020-03-02:
I am not able to reproduce this. I do hear a strange noise on the right speaker during silence from the IO board analog out on this game but nothing with optical audio or HDMI->optical.

WonkerChonker, 2020-03-02:
@sorgelig : All tests were run powering the mister up, then the snes core, then a megaman x rom.

@paulob-nl : I tested this once more today. Mister into receiver via HDMI. Yes there is still a strange buzz. Opted for the japanese rom next. Still a buzz. Tried a different HDMI input (ps4's a home for a ton of movies so I know for sure the input is fine) and there was still a buzz. Plugged Mister into CRT via ypbpr and analog audio out (3.5mm) to receiver resulted in clear audio.

If you meant to look for the buzz, try cranking it up past 75db to around 80-85ish? That helped for me

Honestly this fixes it for me. I can just force the mister to go directly into my TV via HDMI and 3.5mm all the audio over to my receiver. I will probably buy a SPDIF converter to utilize optical out from the IO board soon. Stoked to be using this wonderful core without having to grit my teeth through the audio in some games.

As you were unable to reproduce the issue, it stands to reason it is something flakey with my audio gear. I will be unable to resist the urge to test further. I am likely to use headphones with some HDMI displays to see if there is something up with the signal or my setup requires troubleshooting. Now, while I automatically point the finger at my gear here at home, I want to make a note that I was specifically avoiding this SNES core for this audio bug. I had no issues with GBA/GB/SMS/GEN/TG16/NES/NeoGEO/any arcade core via that same HDMI input. Megaman X is not the only game I heard it in, but I did not record those and they were not as noticeable as highway stage 01 in BGM test. I had a musician friend verify this audio buzz as I do not consider myself an audiophile or a professional listener.

I notice you mentioned testing this with analog out, toslink out, and HDMI converted to toslink? Would it be possible for you to test hdmi directly?

WonkerChonker, 2020-03-02:
I ran headphones directly into the MiSTer IO board and was able to pick up the noise on the right channel. I then ran MiSTer directly into my TV and found the highway/01 buzz again both from speakers and headphone plugged directly into the audio out of the TV. Obviously a sample size of 1 mister, a pair of roms, and a couple of audio devices is not great and the number of variables to play with is many BUT I think its possible there might be an issue with audio over HDMI with the SNES core using certain games.

paulb-nl, 2020-03-03:
That noise on the right channel is probably related. It only happens with the SNES core with some games.

I tried headphones directly in my TV and didn't hear the buzz. Did you try the 96Khz audio setting? I'm not hearing anything wrong with either 48Khz or 96Khz.

WonkerChonker, 2020-03-03:
OK so it does seem to be related to the 48/96khz setting in the ini settings. I came to this conclusion with some more testing. All previous tests were run at 48khz and I intended to leave the system at 48khz permanently just to prevent any sound compatibility issues down the road (I mean for my mister to travel and be hooked into different setups constantly). Tests as follow:

Using an alternate receiver, headphones out of the receiver, 48khz, BGM 01 megaman x
HDMI out resulted in a buzz
Analog out resulted in no buzz
Optical resulted in a buzz
At this point I concluded it was likely an issue with the digital signal.

Using that same receiver, headphones out of the receiver, ini set to 96khz, BGM 01 megaman x
HMDI out NO buzz
Optical out NO buzz
Analog out NO buzz (still has right side high pitched noise/other issue)

On the one hand, I would wager that the core is designed to utilize the higher quality signal. Maybe the INI settings and the core don't talk? 
On the other hand, nobody else seems to be able to recreate the issue. I find this very frustrating. So, to recreate the issue I am having:

Set audio in the .ini to 48khz
Use a digital signal, HDMI or Toslink. (analog will be your control)
Install it directly into a sound device such as a receiver or TV
Use decent headphones or speakers directly out of your receiver or TV. 75-85 db will really make it stand out.
Crappy speakers like those in many modern flatscreen tvs will mask the issue. My main room LCD is 12 years old and the integrated soundbar makes everything sound equally bad. Watch for any kind of middleman device in the chain filtering the audio. 

Let me know if you can recreate it. It's really staring to sound like it is an issue with my MiSTer specifically. Also thanks a million, I can enjoy the SNES core and use a single HDMI to plug the MiSTer into my main receiver again.

paulb-nl, 2020-04-11:
I have tried with 48Khz optical to my receiver with and without headphones and HDMI to my TV with headphones directly to my TV but I don't hear anything wrong.

Are you able to make a recording of the noise?

James-F2, 2020-05-30:
I have zero issues with sound.
Can be closed.

sorgelig, 2020-05-30:
This is really should belong to forum discussion.
Seems specific to setup, not really core issue.

closed 2020-05-30 by sorgelig