MiSTer-devel/SNES_MiSTer #211: Possible combination input problems

edru, 2020-03-02:
Not sure if this is a bug, but I noticed this while playing a Super Mario World Practice ROM http://www.dotsarecool.com/twitch/practice.html

It has options, such as L+R to restart segment, Select + L or R to Save/Load and so on. When using SNAC, these options work fine. When using USB none of them work.

paulb-nl, 2020-03-02:
Working fine here with a USB controller.

edru, 2020-03-03:
@paulb-nl you can do L+R on a usb controller to restart the level? Or Select + R to save and select + L to save load? 

I've tried a usb controller, and it didnt work, then a mayflash usb adapter with snes controller, and it didnt work out.

sorgelig, 2020-03-03:
All mentioned combinations are working fine on USB gamepad. 
Make sure you use non-crippled gamepad or adapters adding some functions to these combinations.

paulb-nl, 2020-03-03:
> you can do L+R on a usb controller to restart the level? Or Select + R to save and select + L to save load?

Yes that all works.

edru, 2020-03-03:
Ill try a 3rd pad. I could see the mayflash adapter throwing dual commands (like LT and LB at the same time). I was doubtful the 8bitdo would be the same way. Ill try a xbox 360 controller and see how that goes. I guess close it up since it is only impacting me. Sorry for the clutter.

closed 2020-03-03 by edru