MiSTer-devel/SNES_MiSTer #212: Incorrect MDFourier output, malformed sync tones

tofukazoo, 2020-03-04:
MDFourier depends on "sync tones", which are a series of audio tones to signify e.g. the start and end of test output. The base sync tone sample on the SNES test consists of two one-frame-long 8khz sine waves, with one frame of silence after each tone. The second tone is out of phase with the first (this is intended). This two-tone sample is played five times, for a total of ten 8khz bursts.

The SNES core is not playing the second half of the two-burst tone correctly, so every other 8khz tone is playing incorrectly, as shown in the attached picture. This should show 10 identical tones:

[image]

Here is a zip file containing both the SNES 240p Test Suite ROM and it's audio output from MDFourier. You can select "MDFourier beta 4" on the main page of the Test Suite to output the test audio:

[240pSuite-SFC_and_Output.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/4284991/240pSuite-SFC_and_Output.zip)

This test plays correctly on original hardware, both the SNS-001 and  SNS-101 models.

gyurco, 2020-06-03:
Is the test tone available in an SPC file? Then it can be checked in the simulator more comfortably than on the FPGA.

sorgelig, 2020-07-02:
should be fixed now

closed 2020-09-02 by tofukazoo