MiSTer-devel/SNES_MiSTer #213: Equinox graphic glitch

Aquas77, 2020-03-30:
Character sprite is clipping into some layers. Only noticed it in this particular room, but fairly positive it is abnormal behavior.

Video: https://www.twitch.tv/videos/578946319

paulb-nl, 2020-03-30:
Any way to get there quickly?

Aquas77, 2020-03-30:
Not that I can find out unfortunately. It's the 4th dungeon and getting to that room is kinda annoying too. The cheats I found none let you skip the dungeons, but there are infinite life and keys cheats, if you felt like breezing through it. I'm doing a full playthrough so I'll report any other issues I find.

paulb-nl, 2020-04-06:
With a save file and some key cheats I could get there easily and I have verified that it happens on a real SNES.

https://youtu.be/RNTP8tl1yYY

Aquas77, 2020-04-07:
Oh, my apologies. I had someone check some real hardware footage and they said it wasn't occurring. I'll try to double verify next time I report. Thanks for checking it out.

closed 2020-04-07 by Aquas77