MiSTer-devel/SNES_MiSTer #214: Broken graphics in Tiny Toon Adventures - Dotabata Daiundoukai (Japan)

Chocobobobo, 2020-04-30:
It affects both US and JP version. You just need to start the game to see this within the first 5 seconds.
Using Core SNES_20200326.rbf
[image]
[image]
 I didn't correct the aspect ratio, but it should be good enough to see the different issues.
Thanks for looking into it!

James-F2, 2020-05-02:
I don't see this problem, everything looks normal.
32MB

Chocobobobo, 2020-05-02:
I have 128 MB, and issue shows up 100% of the time. Can other people check? Thanks.

sorgelig, 2020-05-02:
I see this problem. 
But not sure if it's broken rom or core bug.

paulb-nl, 2020-05-02:
This game seems sensitive to the Initial WRAM setting. 

The newspaper is broken with the 00FF value but it works correctly with the other values.

sorgelig, 2020-05-02:
True!
Issue is solved.

James-F2, 2020-05-03:
Use 9966 WRAM pattern.

Maybe reconsider the 00FF as default, or tweak it a little?

Chocobobobo, 2020-05-03:
Thanks for checking, I could also confirm that 00FF was to blame.
Using this value also made the game crash twice after the third event, so there is definitely something wrong with this combination. Of course, I never had such problem on real hardware.

Sorg, Paulb, what are your thoughts to address this?
-Start a list of WRAM - Game incompatiliby and force a different value when problematic ROM are selected?
-Delete 00FF as a potential WRAM value, or change it? How were those hard values determined? I think Byuu was mentioning something about it, but it was still a mystery IIRC.
-Another option?
Of course, adding more values mean checking the full library to make sure similar issues are not happening.
Thanks for looking into this! Looking forward to read your feedback :)

paulb-nl, 2020-05-03:
The values were determined by reading uninitialized RAM on real SNES consoles. So this bug should happen on some consoles unless 00FF triggers a bug in the core which does not happen on real hardware. Though it seems unlikely that they wouldn't have released a new version if this happens on real consoles.

The game is setting up some DMA transfers by reading from uninitialized RAM so if there is a bug with DMA and invalid values then we could be seeing that.

I think we can change the default to the 9966 value.

SegaSnatcher, 2020-05-29:
Time to close this issue?

Chocobobobo, 2020-05-29:
Why do you want to close the issue? The issue is still there with the latest (0511) version of the core.

Or maybe you found the root of the problem and pushed a commit to fix what happens with 00FF as a WRAM value for this game?
Changing default value to 9966 will NOT fix the root of the problem.

sorgelig, 2020-05-29:
If you want to fix the root of the problem you have to call the developer of this game, haha :)
Developer needs to fix usage of uninitialized memory.

Chocobobobo, 2020-05-29:
Hi Sorg,
Well, being very familiar with Nintendo lotcheck and testing process and having shipped commercial games on every Nintendo hardware since the GBA, it is close to impossible that a game with 2 graphic corruptions (including the very first screen) would pass their certification. Owning the cart back in the days, I don't remember a single instance of having broken graphics.

Are you confident anything related to DMA in the core is bulletproof? No clue on your side what might be worth investigating?

Original game shouldn't be too difficult to get once I can fly back to Japan, I will boot it 50 times on real hardware if you want (so I should hit the 00FF value at some point)

Finally, I am not sure the original developers will pick up the phone if I try to contact them now. :)

paulb-nl, 2020-05-29:
I have confirmed this happens on real hardware. 
https://youtu.be/fon28EmkA8s

I have debugged the game and found that if there is $00FF at memory address $7F9000 then it can be glitched if memory near $7F9480 is also $00FF.

I have forced those memory address to $00FF with the Mario world Jailbreak and then quickly swapped the cartridge with a flash cartridge with Tiny Toons on it and it shows the same broken graphics.

As you can see in the video the uninitialized memory was $00F3 on my SNES and that does not trigger the glitch.

sorgelig, 2020-05-29:
Yeah, there are many games with bugs got released even after thoroughly tested. It's simply impossible to test every single case. 
Many systems were jailbroken through bugs in games. So it's not something extraordinary.
I guess N. had no test bench against initial RAM content. That's why many games are glitching, and that's why core has option for initial RAM filling instead of single hardcoded value. 
Probably it's worth to change default filling value to 9966.

sorgelig, 2020-05-30:
New release uses 9966 fill by default.
And this is not an issue of core.
Considered as solved.

closed 2020-05-30 by sorgelig