MiSTer-devel/SNES_MiSTer #215: Emudetect test rom shows Dog when booting after loading a game.

SegaSnatcher, 2020-05-29:
I'm using SNES 1 ram setting and if I load this test after loading a game it shows the Dog, but if I go straight to the test from a cold boot it shows "Original Hardware" picture.  

Is this a problem with the test rom or the core?

paulb-nl, 2020-05-29:
It is because the core starts with palette memory all zeroes and then it shows the dog. The second time  the palette is filled with whatever the previous rom put there.

James-F2, 2020-05-29:
I thought the memory pattern is reset each time a new rom is loaded. ?

SegaSnatcher, 2020-05-29:
I'm just curious if this also happens on a real SNES or is it just a MiSTer issue?

paulb-nl, 2020-05-29:
> I thought the memory pattern is reset each time a new rom is loaded. ?

Yes but not all memory currently. like the palette ram.

> I'm just curious if this also happens on a real SNES or is it just a MiSTer issue?

It depends what is in the palette ram on cold boot. A real SNES probably does not have all zeros in palette ram on boot

SegaSnatcher, 2020-05-29:
So technically since this isn't acting like a real SNES would should I keep this issue open or no?

sorgelig, 2020-05-30:
fixed by filling a "random" data.

closed 2020-05-30 by sorgelig