MiSTer-devel/SNES_MiSTer #221: Star Ocean Magno Spanish Translation Graphical Errors

aberusugi, 2020-06-18:
The Magno Spanish translation rom doesn't function on the SNES Mister core. MD5 for the patched ROM is d5e4d3ad82936cdc4124e418f9035698. MD5 for the unpatched No-Intro rom used to manually apply patch was d686ba6df942084216393ada009126dc. These MD5's match the ones in the Magno Spanish patch instructions. So I've ruled out that potential issue.

I don't have an everdrive to test on original hardware, maybe someone else can do that for me. In bsnes v115 (final byuu release) it runs just fine. Screenshots attached. It doesn't appear widescreen on my tv, but shows up like that in the screenshot for some reason...

[image]

[image]

sorgelig, 2020-06-18:
most likely - bad header. patches seldom bother to correct the header.

aberusugi, 2020-06-18:
Ah yeah, that makes sense. Thanks for replying.

paulb-nl, 2020-06-27:
The header seems fine and the translation works well in Bsnes and Mesen-S. 

It points towards core issue.

sorgelig, 2020-06-30:
it seems just another bastardized patch for sd2snes like numerous SA1 hacks never intended to work on real HW except sd2snes.

aberusugi, 2020-07-02:
I concur. A bit ago I had tested every single Star Ocean ROM in the smokemonster confirmed everdrive pack, and the Magno translation patch was the only one that had any problems.  I also manually patched the correctly matching MD5 ROM with the Magno patch, and the new rom had the same end-result MD5 as the one in the smokemonster pack, and it created the same problem, so I ruled out an improperly applied patch as the reason. It's the patch itself that must be messing with the header.

sorgelig, 2020-07-02:
Sometimes patches made to promote specific HW. Not that i'm sure, but...

sorgelig, 2020-07-02:
Anyway, it already works.

closed 2020-07-02 by sorgelig