MiSTer-devel/SNES_MiSTer #224: Super Mario RPG save was reset all progress lost

will7046, 2020-07-05:
Using current version, auto save is enabled. My previous Super Mario RPG save file loaded just as normal at startup of the game, I then reloaded the game and all of my progress was lost. I created a new save, it would save and let me reload 3 times in a row, the 4th time it reset all of my progress again. I had used this save since SA1 was implemented with no issues. I had not updated my Mister in about a month, not sure which version I was on previously.

aberusugi, 2020-07-05:
32mb, 64mb, 128mb SDRAM?

Asking because I just recently completed a full playthrough of the game successfully and am using 128mb SDRAM module from misterfpga.co.uk for reference and it passes testing at 140mhz.

will7046, 2020-07-05:
32MB Ram module, thanks

will7046, 2020-07-05:
Tried again today with latest Main mister release, would save and then reload game after about 10 times of doing this in a row it would reset my progress again eventually.

aberusugi, 2020-07-05:
Looks like you got a ball rolling. 

https://github.com/MiSTer-devel/SNES_MiSTer/pull/225

James-F2, 2020-10-02:
Fixed in Release 20200929.
Can be closed.

closed 2020-10-02 by sorgelig