MiSTer-devel/SNES_MiSTer #226: SA-1 3D demo (dforce3000) crash

Kaeril, 2020-07-06:
The SA-1 3D demo from dforce3000 shows off a few 3D models rendered through the SA-1 chip.
Here is a [reference video](https://www.youtube.com/watch?v=42GeYsPGSjM) of it on the original hardware.

It works fine on higan v110 as well.

On the MiSTer core though, as of current core (20200701 at time of writing), the demo will display several glitches and incorrect lines on certain frames, and then will crash after about 30 seconds, displaying an actual error message on screen. Please find attached screenshots below.

If that's of any use, the demo source code is in fact available [here](https://github.com/DocSchoko/SNES-SA-1-demo). Also [here](https://archive.org/download/everdrivepack/%40SNES%20-%20SD2SNES%20-%20Super%20EverDrive%202020-06-22.zip/3%20Special%20Chip%20Games%2FSA-1%2FDemos%20%26%20Tools%2FSA-1%203D%20Demo%20%28PD%29%20dforce3000.sfc) is the .sfc rom for more convenience.

[image]

[image]

James-F2, 2020-10-02:
Note that this demo is PAL region and might have some timing inaccuracies.

sorgelig, 2020-10-02:
How it works in emulator is not so relevant. Different emulators have different bugs and tweaks. It's more important how it works on real HW. Also it's more important how SA1 games work rather some tech demo.