MiSTer-devel/SNES_MiSTer #227: Selecting Save Backup RAM from menu does not show a notification

osaeed18, 2020-07-08:
Hi all, when I select a game, complete a level and the game prompts me to save, I choose the Save Backup RAM option and I do not see a notification on the bottom, top or anywhere on the screen that I chosen the option. 

Can this be added as a feature in a future release? 

Thank you

sorgelig, 2020-07-08:
On I/O board you may see disk LED activity on save.

osaeed18, 2020-07-13:
Hi sorgelig, I think it would be pretty neat to have some save notification.  In the program Retroarch, you can see something as a reference to what can be added as a notification feature to the SNES MiSter core. 

[image]

Just my thoughts. Thank you for your contribution to this core.

closed 2020-07-14 by osaeed18