MiSTer-devel/SNES_MiSTer #228: Megaman X SNES Nointro ...bug?

rykerrk, 2020-07-12:
I don't know if this is possible in the physical game; I've beaten it dozens of times across multiple platforms and never run into this before...

Basic dope pathing, starter stage to chill penguin to spark mandrill (picked up heart container), and immediately after the level post password, or by jamming confirm after password (Unfortunately I was looking away, the boss was dead) I was warped back to starting stage (not possible?), lost my boots (certainly shouldn't have happened) and can't escape stage.  Had to clear it again, went back to Chill Penguin stage and apparently Light stole the boots back because they were still there.  And I could escape the stage.

Not sure if it's reproduce-able, not sure what would have caused this other than some sort of miss-timing somewhere.  All I have is this confused video.  Thoughts?

https://youtu.be/rO2vcxAl_LU

rykerrk, 2020-07-12:
Okay, as I jumped into Armadillo's Stage and died in the first hole (don't ask) and got back on the platform, I charged a shot and went down the incline and I think simultaneously released the charged shot, ran into a peacock and got hit, and the stage faded out and I got the rolling shield weapon.  Something's not right; I'm gonna try rebooting the DE-10 Nano and seeing if I can get a normal play session out of this.

sorgelig, 2020-07-13:
I think it's for forum discussion, not for GitHub.

rykerrk, 2020-07-15:
Okay, thread has been started by someone else having the same problem here

https://misterfpga.org/viewtopic.php?f=18&t=747

I will close this so we can stare at that thread!

closed 2020-07-15 by rykerrk