MiSTer-devel/SNES_MiSTer #236: BSX/Satellaview Zelda game hacks unplayable

Kitrinx, 2020-09-29:
Many games have been hacked to work without integrated BSX support, however many of these hacks fail to load on the core making these games unplayable in any way.

Some games:
BS Zelda no Densetsu - Kamigami no Triforce (can't get past name entry)
BS Zelda no Densetsu Kodai no Sekiban Dai (all of the parts) (wont load)
BS Zelda No Densetsu Remix (scrambled graphics)

srg320, 2020-09-29:
Where can I find these hacks?

paulb-nl, 2020-09-29:
Can you be more specific which hacks you use?

BS Zelda no Densetsu - Kamigami no Triforce (Triforce of the Gods) doesn't seem to have a patch to use it without BSX.

BS Zelda no Densetsu Kodai no Sekiban Dai (Ancient Stone Tablets) Base Patch+Official Indoor Patch+Upgrade patch works or Official Week 1 - REPATCHED 09/21/20 also works

BS Zelda No Densetsu Remix appears to be BS Zelda Restoration Map 1 which works.

Kitrinx, 2020-09-30:
It seems the updated patches for these ROMs run better and work with the core, with the exception of Triforce of the Gods which does not allow name entry, but this seems like a low quality patch, so I will close this issue.

closed 2020-09-30 by Kitrinx