MiSTer-devel/SNES_MiSTer #237: Odd flashing on part of screen while playing super mario all stars + super mario world

sethenstien, 2020-10-04:
At random times while playing super mario all stars + super mario world I get a line of pixels that flash random colors on the right side of the screen. Best picture I have for now is below. Thanks.
[image]

Kitrinx, 2020-10-04:
Please show the equivalent on real hardware, to verify this is not simply game behavior.

paulb-nl, 2020-10-04:
This happens on Mario World when CPU Turbo is enabled.

James-F2, 2020-10-09:
Yep, caused by Turbo.
Issue can be closed, user error.

[image]

closed 2020-10-09 by sorgelig