MiSTer-devel/SNES_MiSTer #239: Don't show screen 'Loading rom' in 15 khz signal

mabrilG, 2020-10-15:
SNES core don't show image when is loading rom in 15 khz signal, some cores like NES do it well. It seems to make a mode change to 31khz during that time. I think it is a minor detail, but changes the signal without having to do it.

Thanks for all the work done to improve this great core.

sorgelig, 2022-05-18:
i'm not quite sure what do you mean by "15Khz signal". Are you using VGA output or custom video mode over HDMI?

Thelypody, 2022-05-18:
Currently, when using an Analog I/O board with a VGA-SCART connection to a consumer CRT TV, the screen blacks out while a rom is loading. This does not happen in many other cores, where a loading bar is visible instead. I can only guess, but I assume that the SNES core generates no video during loading, leading the an image loss on analog display.