MiSTer-devel/SNES_MiSTer #242: Issue loading some .bs files, like Marvelous - Camp Arnold Course - Dai-1-shuu (Japan) (SoundLink).bs

Chocobobobo, 2020-10-17:
The title says it all. Here is what we have currently (using the patched US Bios)

[image]

Please note that the exact same .bs file works fine on SD2SNES.

Thanks for looking into it!

srg320, 2020-10-18:
Some BS roms contain incorrect values in the header. 
This will be fixed in Main in next release.

Chocobobobo, 2020-10-18:
Just saw your commit, thanks a lot for the quick turnaround.
I will do more testing once I could check your fix.

Chocobobobo, 2020-10-18:
Fix confirmed, thank you.

[image]
[image]

closed 2020-10-18 by Chocobobobo