MiSTer-devel/SNES_MiSTer #243: BS-X: "Kirby no Omochabako - Baseball" and "Kirby no Omochabako - Pinball" don't boot

TLPD-AVW, 2020-10-17:
Two games from Kirby' Toy Box series - "Kirby no Omochabako - Baseball" and "Kirby no Omochabako - Pinball" refuse to load through BS-X English DRM-free BIOS. Trying to load either of them produces Error 09.
[image]
[image]
Five other games from that series do not suffer from this issue and load properly. These two .bs roms run properly with the same BIOS on Snes9x.

srg320, 2020-10-18:
Some BS roms contain incorrect values in the header. 
This will be fixed in Main in next release.

Chocobobobo, 2020-10-18:
Both are fixed after your earlier commit, thanks srg320.

[image]
[image]

srg320, 2020-10-25:
Both games boot fine for me, tested several times.

Chocobobobo, 2020-10-26:
Pinball is fine now, just checked with your snes: fix has_bsx_slot init value commit.

TLPD-AVW, 2020-10-27:
Both games start properly with the latest releases of Main and SNES core. I'm closing the issue, thanks a lot for your work!

closed 2020-10-27 by TLPD-AVW