MiSTer-devel/SNES_MiSTer #244: Two more .bs files not being recognized, even after this fix (* snes: fix patch of incorrect BS header values)

Chocobobobo, 2020-10-19:
First of all, please note that those tests have been done using an updated MiSTer and SNES including your commit (* snes: fix patch of incorrect BS header values)

I found two more .bs files not booting using the default japanese BIOS, but are working fine on bsnes for example.

Those two games are:
Spriggan Powered BS - Dai-3-wa (Japan) (SoundLink)
Spriggan Powered BS - Prelude (Japan) (SoundLink)

Thanks for looking into it.
Right now, this is what we have:

[image]

Chocobobobo, 2020-10-19:
Extra info

Spriggan Powered BS - Prelude (Japan) (SoundLink) loads with the hacked US BIOS, but nothing with the original BIOS.

So there is still something going, based on the BIOS that you are using. (https://github.com/MiSTer-devel/SNES_MiSTer/issues/241)

srg320, 2020-10-25:
Both games load fine with japanese bios.
[image]
[image]

Chocobobobo, 2020-10-26:
Not sure how this is related, but I rechecked after your earlier commit snes: fix has_bsx_slot init value, and it is fine now, thanks srg320 !

closed 2020-10-26 by Chocobobobo