MiSTer-devel/SNES_MiSTer #247: Memory Pak used by some games

wwark, 2020-10-27:
During the launch of the game Sound Novel Tsukuru, we have a message indicating no Memory pak presents.

I found this in higan documentation https://higan.readthedocs.io/en/stable/guides/import/ (satellaview section)

The control cartridge had a slot that accepted re-writable “memory paks”, so that people could store the games and data they downloaded. **A small number of games that did not use the Satellaview modem also had a memory pak slot**, so the game’s publishers could publish extra content for the game via the Satellaview service after the game’s release. For the benefit of people who didn’t own a Satellaview some read-only memory paks were sold in retail stores containing extra content for specific games.

I tried Sound Novel Tsukuru on Higan and we have panel asking BS memory before the game loading.
I don't have the list of games required this memory paks but maybe this list is present on Higan code.

Thanks,

wwark, 2020-10-28:
The picture of the cartrige of Sound Novel Tsukuru
[image]

wwark, 2020-10-28:
I found the list of games : https://wiki.superfamicom.org/data-packs

birdybro, 2021-04-10:
The memory pak feature isn't even supported by Higan. There is a fork of bsnes here --> https://project.satellaview.org/downloads.htm which supports it, but takes a little figuring out to make it work. You have to load special. The source code is available.