MiSTer-devel/SNES_MiSTer #248: image has stutter/ tearing SNES_20201027.rbf and latest mini, not in SNES_20201014 with previous main.

Psygho, 2020-10-29:
I used SNES_20201014, with the then latest main, No stutter in Mario or other platform games. Used standard ini file, with Vsync adjust =0

When upgraded to SNES_20201027.rbf, (and mandatory latest main), i get stutter, and tearing in game play. Most noticible in SMW. In other cores no problem, just in Snes core.

When used option : vsync_adjust=2 , stutter went away in latest core.

Treid to manualy go back to  SNES_20201014, but with latest main, the core crashed when try to load a rom.

Do not know how to rol back to previous MAIN , so could not test if that would solved the problem.

Dont know it has to do with the latest main, or the latest SNES core though.

paulb-nl, 2020-10-29:
I have tested SNES 1027 with Main 1027 and I didn't see any problems with vsync_adjust=0 or vsync_adjust=2. 

You should be using vsync_adjust=2 anyway because that is the only one with zero lag.

Psygho, 2020-10-29:
Got latest main and snec core with update....
I manualy download and copied latest snes core back on SD card, and now it is working fine???
Can not repplicate error again.  Strange because i could replicate it over and over again..

Maybee a glitch ? Dont know.

**Thnx for support, and taking is serieus!**

I know i should use vsync2, if possible...  But for testing i had it on mine cheap ass monitor, instead of mine oled c7v in the living room....

closed 2020-10-29 by Psygho