MiSTer-devel/SNES_MiSTer #249: Ogre Battle - The March of the Black Queen - Sprite issues

wwark, 2020-11-01:
It is in the beginning of the game after start a new game (visible on USA version and Japanese version)

[image]
[image]
 
Thanks,

paulb-nl, 2020-11-02:
There are 2 Japanese versions.
Normal: Size: 1536 KB, CRC32: 65CD17E3
Nintendo Power: Size 2048 KB, CRC32 3E50C7C2. This one has an extra "Load" option between New game and continue.

I can only get the green tiles to appear by loading the Japanese Nintendo Power version. After that the English version will also have the green tiles.

Reloading the core does not fix it. Only rebooting to the menu and then reloading the core works.

wwark, 2020-11-02:
Thank you @paulb-nl.
I confirm the issue happens if we launched Japanese Nintendo Power Version first.

wwark, 2020-11-14:
Fixed by this commit (https://github.com/MiSTer-devel/SNES_MiSTer/commit/57c00ce575e845be7b7ca56c11e7550240a0df49)
Thanks @srg320 !

wwark, 2020-11-16:
For information (maybe related to my build), I have some issue to launch BS games.
After select the gender, the core becomes very slow (hang).
I don't have this behaviour with the last release RBF.

wwark, 2020-11-23:
Oups I forgot to reopen this issue.
Do you think it is link to my compilation or is it a regression ?
Thanks,

wwark, 2021-01-21:
Last update Sattelaview working.
Thanks !

closed 2021-01-21 by wwark