MiSTer-devel/SNES_MiSTer #251: Screen flickering issue when initiating dialogue

2EPaladin, 2020-11-24:
I've noticed a screen flickering issue from at least a few versions back that is still currently happening. Whenever I play Treasure of the Rudras or Crystal Beans I get a screen flickering issue that happens whenever a character opens a new dialogue box. If a person has a few boxes of dialogue the screen will be fine till the next person starts and then the whole screen flickers for a brief moment before returning to normal. I've noticed this since I've started using my MiSTer a few months ago and it is definitely still in the newest core. Whether this affects more games though I'm not sure as I haven't had time to extensively test.

As for things I have already tried for troubleshooting: 

Switching the SNES modes between the models and turbo.
Tried setting a min and maximum refresh rate, though I haven't extensively messed with this
Messed with the max resolution and each scaling option to see if one of these would fix the issue
Tried it on a different display albeit by the same input option (HDMI)
Tried finding a different ROM and applying the English patch myself
Played the unaltered Japanese ROM

All of these still displayed the same issue. Playing the game in SNES9x fixes this issue, but if I wanted to do that I wouldn't have built a MiSTer. Tried posting about this on fpgagaming to see if anyone had a similar issue but they thought it might be more fruitful to try here.

sorgelig, 2021-03-08:
Some games may dynamically switch resolution when dialog appears.
It's not an issue, it's normal work.

closed 2021-03-08 by sorgelig

paulb-nl, 2021-03-09:
This flicker only happens when Scandoubler is enabled because without scandoubler the resolution is always 512px and with scandoubler it changes from 256 to 512 with this game.

I think it is done this way just for better smoothness with HQ2x? 

There should be an option to always keep 512px so that the flicker doesn't happen or it should only switch between 256 and 512 if HQ2x is used so that normal Scandoubler is not affected.

sorgelig, 2021-03-10:
Disable scandoubler and set Force 256px = Off.
It will use 512px always.

paulb-nl, 2021-03-10:
Yeah if people don't need Scandoubler then its possible but if they use HQ2x or HDMI+VGA monitor with force_scandoubler=1 then there will be this flicker.

It would be better to change "Force 256px,Off,On" to "Force resolution,Off,256px,512px"

sorgelig, 2021-03-10:
HQ2x won't work with wrong pixel clock.