MiSTer-devel/SNES_MiSTer #253: Super Mario World (USA) - Graphical glitch

bibou42, 2020-12-31:
Hi!
While fighting the boss in Ludwig's Castle, I noticed some graphical glitches (see picture).
I can't test on real hardware but the glitch doesn't happen in bsnes emulator.

[image]

Max833, 2020-12-31:
Maybe that is because of that
https://github.com/mrehkopf/sd2snes/issues/126

James-F2, 2021-03-09:
That's normal on real console.

sorgelig, 2021-03-09:
@James-F2 thanks for checking it!

bibou42, 2021-03-09:
Thanks for checking :)

paulb-nl, 2021-03-09:
@James-F2 Did you only try it with an SD2SNES?

I have used a real Super Mario World cart on a APU-01 console and there are no glitches at all while MiSTer shows a lot of glitches when Ludwig is rotating or standing at the right edge.

APU-01:
https://youtu.be/rdufK1PF8eg

MiSTer:
https://youtu.be/WxhqSj8h76E

bibou42, 2021-06-21:
Fixed in Release 20210620

closed 2021-06-21 by bibou42