MiSTer-devel/SNES_MiSTer #255: Analog video regression with latest release

henryrizzi, 2021-01-21:
In the latest release of the core at the time of writing (20201106), along with a build of the most recent commits, there is a regression in analog video output. The issue usually manifests as light green dots scattered through the video output, but one of the most easily replicated examples of the graphical issue is within the Color Test portion of the [SNES Test Program](https://tcrf.net/SNES_Test_Program).
[image]
I've attached an image with the line glitch circled via RGBHV output from the analog board. I was able to replicate the same line glitch behavior across all the main analog video options (RGBHV, RGBs, YPbPr), but the issue is not present when using direct_video output. 
The regression occurred between build 20201027 and now, so it appears to be related to system changes for custom aspect ratio support.

sorgelig, 2021-01-21:
try this build:
[SNES.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/5847462/SNES.zip)

henryrizzi, 2021-01-21:
Hello Sorgelig! That build resolved the issue in the SNES test program (along with the most recent release).

I realized there is another, separate issue, with light green pixel artifacting which affects games such as U.N. Squadron (there are light green specs in the mountains and ground area on the first stage via analog but not in HDMI output). But this does not appear to be a regression from the same version jump (and may have been present for a while). I'll try and find anything more obvious in testing, but I really appreciate you tweaking the system video out for analog.

Thank you!

closed 2021-01-21 by henryrizzi