MiSTer-devel/SNES_MiSTer #259: Race Drivin SA1 Hack does not run accurately.

SegaSnatcher, 2021-02-22:
This is a new SA1 hack from Vitor and it appears MiSter can not accurately run this hack like a FXPak in a real SNES or software emulators like Snes9x and bsnes/higan/ares can.  

One noticeable flaw with how MiSTer runs it is that the RPM dial goes wacky and moves around randomly.  

Below I will show comparisons letting the game run in demo mode. 

Here is the hack running correctly.

https://www.youtube.com/watch?v=ThMRXZbiIcE

This is how it runs on MiSTer. (Notice the RPM needle and the fact the car crashes, the car is not supposed to crash.)  

https://www.youtube.com/watch?v=I0EByc2Td9g&feature=youtu.be

Here is the link to Vitor's SA1 hack github, might have useful information.

https://github.com/VitorVilela7/SA1-Root/tree/master/Race-Drivin

Max833, 2021-02-22:
This could be helpful as well: https://github.com/VitorVilela7/SnesSpeedTest/releases/tag/v5.1

paulb-nl, 2021-02-27:
The game has the same issues on a real SA-1 cart so actually the MiSTer core is the only accurate one. 

https://youtu.be/X9DUeS0r5ZE

SegaSnatcher, 2021-02-27:
> The game has the same issues on a real SA-1 cart so actually the MiSTer core is the only accurate one.
> 
> https://youtu.be/X9DUeS0r5ZE

Do you think 3-chip vs 1-chip behavior has anything to do with this.   Very odd that the issues are found on a real SA1 modded cartridge as well.  I guess this is good news for SA1 implementation on MiSTer.   I should show this to Vitor.  

Thanks for posting this.

SegaSnatcher, 2021-02-27:
Vitor made a comment about this, it looks like he never tested using a real SA1 cartridge. 

"I couldn't test Race Drivin' on real SA-1 cart because of the renovations I'm currently doing on my house

The RPM display is bugged likely because the SA-1 CPU and SNES CPU aggressively runs in parallel & the SNES CPU does read while SA-1 CPU is doing calculations on the RPM counter.

The fix is simple and involves using a stable memory address which is only stored once when done."  

So, good news is this probably isn't a problem with MiSTer.  I'll keep this open until we can confirm that a future update from Vitor fixes these issues.

closed 2021-03-08 by sorgelig