MiSTer-devel/SNES_MiSTer #261: Beavis and Butthead anti-piracy still in effect

soullos, 2021-03-05:
Beavis and Butthead has what I understand an anti-piracy effect where any point of damage will result in an instant death. I use to own the original cart and it was never like that. Normally you'd lose a bit of life from your life bar when getting damaged. To reproduce it, just load up any level and get hit. It's one hit kills so pretty easy to reproduce.

A few users on the MiSTer discord said it might be a case of a memory read issue. I've tried changing the "Initial WRAM" options in the Hardware menu in the core but neither option works. I've switched to each option, reset the game, and that had no effect. I've tried switching to each option during a single play through after each death, but that likewise had no effect. Interestingly, when I load the game in Retroarch on my PC, the one hit kill effect is gone, you take damage as you should. Not quite sure what it could be, just bringing attention to it. Let me know if anyone needs more info and whatnot.

Thanks. :)

paulb-nl, 2021-03-06:
The game does a SBC $0200 in decimal mode to subtract 1 from the life bar but the MiSTer core calculates the wrong result. With Mesen-S the result of $0B - $01 in decimal mode is $0A but here the result is $94.

If I enable a cheat that sets the $0222 value to $09 and then disable the cheat then it subtracts correctly (9->0->99) and the life bar works as it should.

[image]
[image]

paulb-nl, 2021-03-08:
Fixed by 5323b12f4cc95eac7220befc4e6688c589744d4d Thanks @srg320

birdybro, 2021-04-10:
Current release passes test_adc and test_adc_16, but fails test_sbc and test_sbc_16.

After the recent change this is the opposite. Is the Beavis and Butthead anti-piracy also relying upon functioning adc behavior concurrent with these test roms?

[snes_adc_sbc.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6290570/snes_adc_sbc.zip)

birdybro, 2021-04-23:
Latest update restores ADC test functionality and retains the new SBC fixes!

Bravo @srg320 👏🏻 !!

thehughhefner, 2022-05-06:
I believe this issue can be closed

closed 2022-05-08 by sorgelig