MiSTer-devel/SNES_MiSTer #264: Add De-Jitter to Analogue Output

mirkorean, 2021-03-26:
I know that the core is behaving exactly like the SNES, but since the NES core has a de-jitter in place, i was wondering if we can have it for the SNES (if not as a default, maybe as an option) as well.

Background:
I was trying to integrate the MiSTer into my capture setup (MiSTer -> Analogue SCART -> SCART Switch -> OSSC -> Capture Card (Magewell USB)). The OSSC and/or the capture card have problems with the SNES core (losing sync, switching from 60hz to 120hz). I'm guessing because of the 'jittery' sync signal of the SNES/NES.

mirkorean, 2021-03-28:
NES implementation for reference:
https://github.com/MiSTer-devel/NES_MiSTer/blob/44e592cdc776b0cc7da9678bdcd6faa6ef959765/rtl/nes.v#L191

birdybro, 2021-04-10:
Analogue SCART<-- what is this component specifically? Are you using the Analog I/O board in conjunction with an RGBHV VGA cable --> SCART which has been appropriately attenuated for the MiSTer output, or is this an **Analogue Inc.** product, or is it the Retro-Access Analogue NT Mini cable (which isn't directly compatible with the MiSTer VGA port), etc...?

mirkorean, 2021-04-11:
Thanks for question, sorry if that was confusing. I‘m using the Mister I/O board with the VGA port and a VGA to RGB Scart cable i bought from a european Mister seller. Every other core i tested works fine with this capture setup so i‘m pretty sure its the jitter.

birdybro, 2021-04-11:
Have you used the magewell usb capture utility to adjust your capture card's settings? Just curious to try and help you resolve it in case your proposed change isn't implemented :)

paulb-nl, 2021-04-11:
The OSSC shouldn't have problems with the SNES jitter because it also supports an original SNES. 

It does have problems with SNES if the scandoubler is enabled. Then the OSSC loses sync constantly.

Check that H-PLL Pre and Post-coast are still at 1 and 0 respectively.

mirkorean, 2021-04-11:
H-PLL Pre and Post-coast are still at 1 and 0. I guess the problem is not the OSSC but the Magewell Capture Card. It's just that NES is working fine with every mode (including 5x). The Magewell is even doing perfect with every weird Deinterlace Mode from the OSSC (coming from a 480i Wii) but if i switch to SNES (tested via original SNES or Mister Core) there is no mode working and its going crazy. I tried a bit with the magewell settings but nothing seems to improve the situation. I was curious why the NES core would work then so i asked in the discord and @Kitrinx told me about her fix for the NES core.

birdybro, 2021-04-11:
I just tested around with my USB Magewell Gen1.

The Jitter goes away when you change Video Format from XRGB to default in OBS (indicating that the signal from the SNES is not XRGB or not being detected as such). YUY2 eliminates the jitter for me.

Video example (no jitter) --> [2021-04-11 10-29-08.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6292250/2021-04-11.10-29-08.zip)  
MiSTer settings - 1920x1080, match output frequency, integer scaled, SNES Interpolation filter.  
Magewell settings in obs - default  

Video example (jitter) --> [2021-04-11 10-30-31.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6292254/2021-04-11.10-30-31.zip)
MiSTer settings - 1920x1080, match output frequency, integer scaled, SNES Interpolation filter.  
Magewell settings in obs - changed only video format from default to XRGB

So this seems to indicate that the problem is in the Magewell or that the MiSTer SNES core might be sending a signal that the Magewell and other capture cards might be having an issue with in terms of Video format (and YUY2 is the only thing that resolves it).

mirkorean, 2021-04-11:
Hey @birdybro, i appreciate the help. I think HDMI from Mister would work because i guess it "normalizes" the signal with a buffer, but that's just not working with my setup currently. With direct video and the retrotink 2x (if its just linedoubling without a buffer) i think you would have the same problem as me => no steady picture for even a second with the magewell.

birdybro, 2021-04-11:
Have you switched from XRGB to YUY2 like I mentioned?

mirkorean, 2021-04-11:
Did not change anyhing for me.

birdybro, 2021-04-11:
Okay, that's good to at least try that. Thanks.

birdybro, 2021-04-11:
What's the Magewell capture settings in OBS currently?

mirkorean, 2021-04-11:
I have everything on Device Default (changed the Video Format to YUY2 to try), which works fine for everything besides the SNES core.

birdybro, 2021-04-11:
Do you have a video of the behavior, I'm trying to reproduce it with my relative setup just to see. Kinda curious.

mirkorean, 2021-04-11:
https://www.youtube.com/watch?v=KO-HxxN_TIs

birdybro, 2021-04-19:
Oh wow, that's bizarre. Thanks for sharing the video, hopefully it helps troubleshoot the issue.

Sick setup behind you! :)

sorgelig, 2021-04-19:
Some games are very sensitive to timings. So core's task is to replicate everything precisely. There is no such task as "adjust the core for video capture device". It's video capturer's task to be tolerant to small imperfection as a nature of some devices.

Kitrinx, 2021-04-19:
NES core only works because I made sure the entire core could be paused to make the video timings even. SNES core was unfortunately not written with this capability. The amount of work to add this would be enormous. If you tried to make video timings even without pausing the core, the result would be quite bad in terms of accuracy and stability.

Maybe if someone revisits this core in the future they will take the huge amount of time to add pausing for even frames and for save states, but until such time, it is what it is.

mirkorean, 2021-04-19:
Yeah, i only opened this, because i saw it being done in the NES core. Thanks @Kitrinx for the explanation. All good.

bootsector, 2021-05-03:
Following up on this because I think this is the reason I get a slightly "shaking" image when using scandoubler on my CRT VGA monitor with SNES core and not with NES core.