MiSTer-devel/SNES_MiSTer #265: Add support for XBAND cart

tjanas, 2021-04-09:
Even though the online service is unusable, the XBAND cart itself has a hidden snake maze game that can be accessed from the main six icon screen by pressing Down, Down, Left, Left, Right, B. It would be neat if it was accessible via SNES_MiSTer

birdybro, 2021-04-10:
Not even bsnes loads the XBAND rom currently. There is only one emulator that can even run the XBAND rom, and it's a fork of bsnes called bsnes-sx2, and I never got it to work with that either.

tjanas, 2021-04-11:
The cart also works as a nice test of the XBAND keyboard. :)

Kitrinx, 2021-05-03:
This is actually completely useless. Only a very small number of games supported this at all, even when the service was active. Even if you managed to emulate the original network's modem functionality, the variable latency inherent in modern networks and the heavy use of NATs will make it extremely difficult for people to connect to each other and then have a good experience.

The original used a low speed modem to avoid any kind of jitter, modern networks, while much higher bandwidth, will likely have more jitter than those low speed modems would.

Even if you did emulate the dial up, and did overcome any technical limitations associated with this, and did add a way to snap carts on over the xband rom, and added all it's hardware and keyboard support, then you have a handful of games that never had an ideal multiplayer experience to begin with. Why would you need this feature for anything?

This would honestly just be a complete waste of fpga fabric and best kept in the history books. If you want multiplayer, consider a more modern viable ways of dealing with it. Maybe some emulator author will have time to burn someday and implement it for preservation purposes, but it has no practical uses.

tjanas, 2021-05-03:
I'm just looking at it in terms of preservation, as it's one of the primary goals of the MiSTer project (many of the implemented computer cores have no practical purpose either). There's an active [project](https://xband.retrocomputing.network/) with [reimplementation of the XBAND server](https://twitter.com/agirisan/status/1268715986531897344?s=20) and some of the XBAND features besides 1-on-1 game matches have become functional again. The SNES XBAND cart was [dumped](https://datomatic.no-intro.org/index.php?page=show_record&s=49&n=3363) a couple years ago.

Kitrinx, 2021-05-03:
Preserving what? A thing that has no practical purpose and takes up resources? There will never be another time in which the xband is useful for anything. It was progressive idea for it's era, but ultimately we now live in an era in which that technology will never be useful again. For the purposes of preservation, it would be equally effective to write the HDL and store it unused in a github to maintain the logic of it. 

Take a step back and look at what you're asking for.