MiSTer-devel/SNES_MiSTer #268: SPC Player has incorrect "initial register values"

birdybro, 2021-04-11:
I believe the audio bugs I've observed in multiple SPC files when using the SPC player might be related to it not passing this test.

[initial_regs.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6292702/initial_regs.zip)

From the readme for the test:

```
SNES SPC file loader tests
--------------------------
These verify basic operation of an SPC player on SNES hardware.

They report the result via a series of beeps:

	Beeps			Result
	- - - - - - - - - - - - - - - - - - - - - - - - -
	low				Passed
	low high		General failure
	low high low	Failure code 2
	low high high	Failure code 3
	etc.

Only the first failure is reported. Sub-tests are run in the order of their codes. Once a test is done, it re-runs the bootloader, so the SNES doesn't have to be reset before loading another SPC file.

initial_regs.spc
----------------
Verifies initial register values. Failure codes:

	2: CPU registers
	3: $F0-$FF registers
	4: DSP registers
```

The behavior for multiple SPCs is not functioning currently, and it's all tied into the initial start of the song. Actraiser SPC's fail to go beyond the first split second of playback. Some Chrono Trigger songs have messed up audio for the first sample, etc...

[ActRaiser.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6292705/ActRaiser.zip)

[Chrono Trigger.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6292707/Chrono.Trigger.zip)

For some examples.

thehughhefner, 2022-01-02:
As of v211214 I get Failure code 3.

birdybro, 2022-05-25:
Resolved with https://github.com/MiSTer-devel/SNES_MiSTer/pull/334

closed 2022-05-25 by birdybro