MiSTer-devel/SNES_MiSTer #274: Mode 7 M7VOFS latching test (for reference)

paulb-nl, 2021-05-06:
I have modified the mode 7 tutorial from https://wiki.superfamicom.org/mode-7-scaling to write to M7VOFS in an HV-IRQ to see when the write takes effect.

If the write is on time then there is a black line above the second row of colored squares like so:
[image]

There are different versions of the test slowrom/fastrom and bra and wai. The bra ones wait for the interrupt in an infinite loop.
````
-:
lda $10
bra -
````
[mode7_tests.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/6435237/mode7_tests.zip)

The test starts with interrupt H: 0, V:36. Press the L/R buttons to decrement/increment the H value. Press select to reset everything.
Below are the results. The number is the value of the H-IRQ.

SNES APU-01, 1-CHIP-01 with Everdrive:
````
wai_nops83           1 flicker
wai_nops95_fastrom   3  flicker
wai_nops96_fastrom   1 is too late (no flicker)
bra_nops82           1-6 flicker
bra_nops83           0-3 flicker
bra_nops95_fastrom   1-5 flicker
bra_nops96_fastrom   0-2 flicker
````

MiSTer with commit a04ac87bbaa242e0558f561fe0e2bdfda17be9b9:
````
wai_nops83           4 flicker
wai_nops95_fastrom   6 flicker
wai_nops96_fastrom   3 flicker
bra_nops82           6-10 flicker
bra_nops83           2-7 flicker
bra_nops95_fastrom   4-9 flicker
bra_nops96_fastrom   1-6 flicker
````

Here is an example video of bra_nops_82 on an APU-01 SNES. It starts with the write on time and then I press R 7 times until it stops flickering so the line flickers at H-IRQ 1-6.

https://user-images.githubusercontent.com/2227984/117311446-7b26f980-ae84-11eb-8f11-3fc843fd2943.mov

srg320, 2021-05-07:
Thanks @paulb-nl.
With your tests I determined the right time by latching Mode 7 registers, which is H_CNT = 11. With this time also no glitches in Super Mario World. 

I'm not a fan of adjusting the values for specific games/tests, but I don't see any other option at the moment.

paulb-nl, 2021-05-07:
Thanks @srg320.  The timings are very close now. These are the results with latching at H_CNT 11:
````
wai_nops83          1 flicker
wai_nops95_fastrom  3  flicker
wai_nops96_fastrom  0 flicker
bra_nops82           3-7 flicker
bra_nops83           0-4 flicker
bra_nops95_fastrom  1-6 flicker
bra_nops96_fastrom  0-3 flicker
````

paulb-nl, 2025-01-19:
After your SCPU accuracy fixes on 1 Oct it was almost the same as real hardware. Only bra_nops82 had 2-6 flicker instead of 1-6, the rest was perfect.

Now after commit  65C816: fix IRQ interrupt (emudetect) 657758e9f267670583e64def167b317282fd5c02 there are big differences again.
bra_nops82           1-5 flicker should be 1-6
bra_nops83           0-1 flicker should be 0-3
bra_nops95_fastrom   0-3 flicker should be 1-5
bra_nops96_fastrom   0 flicker should be 0-2

paulb-nl, 2025-02-14:
Thank you. If your hardware has 2-6 flicker then it is perfect now.

closed 2025-02-14 by paulb-nl

Max833, 2026-09-19:
It looks like when this is correctly implemented, it fixes https://github.com/MiSTer-devel/SNES_MiSTer/issues/253 (glitches in Super Mario World Ludwig's Castle boss fight)  and https://github.com/bsnes-emu/bsnes/issues/397 (Jurassic Park's fourth attract sequence, no issues with Mister and in Mesen-CE this is now fixed as well).
Thanks paulb-nl!!