MiSTer-devel/SNES_MiSTer #276: Communication over USB like SD2SNES

mickednk, 2021-05-07:
This is a feature request for implementation of the USB communication protocol used by the modified firmware for SD2SNES/FX PAK PRO (https://github.com/RedGuyyyy/sd2snes)

This is mostly used for auto-tracking inventory in randomizer games and also to do multiworld sessions of Zelda, Super Metroid and combo randomizers. I can live without auto-tracking but multiworld sessions is a no go without this.

I understand if this is not high priority but would be nice to know if it is possible to implement and if it is on your radar for future features.

radon0303, 2021-10-14:
Would love to see this implemented too. The ability to read what's in memory onto your PC would be amazing.

mrchrislad, 2022-04-05:
I'll add my voice to this being on my wish list if anybody can help/do it!

radon0303, 2022-04-05:
This has been brought up as a general issue in the main core as well. See MiSTer-devel/Main_MiSTer#492

Altazimuth, 2022-10-22:
https://github.com/MiSTer-devel/Main_MiSTer/issues/492 got closed a couple weeks ago as it was considered not possible. How does that impact this and #322? In #322 it's mentioned that it _might_ be doable. Is this still the case, or is it strictly not possible?

beefgrits, 2025-09-05:
I know this is old and probably won't ever happen, but I haven't seen anyone ask about the feasibility of just using snac for the usb port so it still lives in the snes core side and you don't need to copy anything to the linux side

sorgelig, 2025-09-08:
SNAC port cannot handle USB protocol

sorgelig, 2025-09-08:
This feature doesn't really fit MiSTer goal. 
Implementing side features like this is still possible if someone addicted to this will implement it and it will fit nicely to current code.