MiSTer-devel/SNES_MiSTer #279: Full Throttle - All-American Racing (USA)

paulb-nl, 2021-05-08:
The bottom line on the water bike race has garbage.

MiSTer:
[image]

SNES:
[image]

There are multiple issues. First issue is that H-IRQ at line 143 is too early. The game writes $0104 (260) to `HTIME`  and then changes from V-IRQ to H-IRQ. Now `IRQ_N` goes low after the switch from V-IRQ to H-IRQ because of `HIRQ_VALID` = 1. The game uses WAI to wait until H 260 but `IRQ_N` is low so immediately exits WAI at H 226.

Here $11 is written to $4200 and `P65_IRQ_N` goes low too early:
[image]

WAI exits too early:
[image]

Second issue is an HDMA issue. The game enables HDMA in the beginning of the H-IRQ at H 260 on line 143. The write to $420C happens around H275-277 probably. On a real SNES the write seems to be too late for HDMA on line 143 so HDMA transfers start at line 144. `NTRL` (Line counter) of HDMA channel 0 should start with $D0. This is also what Mesen-S does. 

The developers of the game chose a very tight timing. If I patch the rom to change the H-IRQ only one pixel to 259 then the background is constantly moving 1 line up and down on a real SNES.

Here I modified the core and game so that the write happens too late for HDMA on line 143 but the core doesn't start the transfer on line 144. It just decrements the Line counter to $CF. Then it starts to write the data from $148000 on line 145 instead of 144 and $148002 on line 146 instead of 145 etc.
[image]

It almost looks correct but if you look very closely the last line is missing:
[image]

MP2E, 2024-10-02:
Just tested and I think this is fixed in the latest changes, the bottom line appears to render correctly now

paulb-nl, 2024-10-02:
Great work by @srg320. Works perfectly with latest changes. Even the patched game with H-IRQ 1 cycle earlier matches original hardware behavior with the background moving 1 line up and down.

closed 2024-10-02 by paulb-nl