MiSTer-devel/SNES_MiSTer #280: Feature Request: Sufami Turbo Support

ajgowans, 2021-05-10:
Can native Sufami Turbo support be added into the core?  I have done an overview of the add on on the forum if anyone wants an overview of the hardware, and to see what it looks like:

https://misterfpga.org/viewtopic.php?f=18&p=25439#p25439

**Sufami Turbo Games**

There were 13 carts released for it:
Poi Poi Ninja
SD Gundam: Generations Part 1
SD Gundam: Generations Part 2
SD Gundam: Generations Part 3
SD Gundam: Generations Part 4
SD Gundam: Generations Part 5
SD Gundam: Generations Part 6
SD Ultra Battle: Ultraman Legend
SD Ultra Battle: Seven Legend
Crayon Shin Chan
Gegege No Kitarou
Bishoujo Senshi Sailor Moon Sailor Stars: Fuwa Fuwa Panic 2
Gekisō Sentai Car Ranger: Zenkai! Racer Senshi

Here is some information on the special features you unlock by having a cart in Slot B:

**SD Gundam Generations**
SD Gundam Generations is a series of 6 games, each containing a certain amount of unique Gundam fighters. With the main Gundam Generations game in slot 1, the player can use Gundam fighters from the cartridge in slot 2.

**SD Ultra Battle**
SD Ultra Battle is a beat 'em up game based on the Ultraman series which allows the player to use the characters from whichever cart was in slot 2, in the main game in slot 1. It links with other copies of the same game (Battle 1 links with Battle 1), allowing both players to use the same character and saved data, or with the opposing game using characters from both games. The first game is based on Ultraman, and the second is based on Ultra Seven.

**Poi Poi Ninja**
Poi Poi Ninja only links with other copies of Poi Poi Ninja, to share save data or character data for vs. battles.

**Implementing Into MiSTer Core**

I am not a developer, but I have been thinking at a high level how support could be added into the SNES MiSTer core.

Firstly have a sufamiturbo folder within the folder structure, containing the BIOS and roms of the 13 carts.
Add a new Sufami Turbo option on the core menu
Selecting this option would load a screen that has options to select rom for Slot A, select rom for Slot B, and launch Sufami Turbo
Launching Sufami Turbo without no carts selected would just launch the BIOS to get the no carts screen and music (if no correctly named BIOS then throw up an error messaging for missing BIOS)
Launching Sufami Turbo with valid rom selected in Slot A would launch the game selected
Launching Sufami Turbo with valid rom selected in Slot A and Slot B would launch rom set in Slot A and apply the additional feature of rom in Slot B

If it were to be developed in stages:
1. Get the Sufami Turbo BIOS loading
2. Be able to load a valid rom
3. Be able to load a valid rom and apply features from valid rom in Slot B

ajgowans, 2021-07-13:
Someone on NES Dev has dug into the hardware and drawn up a schematic:

http://forums.nesdev.com/viewtopic.php?t=19787

tjanas, 2023-11-26:
Support for dumping Sufami Turbo cartridges has been recently added to sanni:

https://github.com/sanni/cartreader/pull/871

sorgelig, 2023-11-27:
Generally speaking, it's possible. But adding manager just for couple niche games looks unreasonable. 
Probably BIOS+CartA+CartB can be merged in one file and handle it in existing compatible mapper

bikerspade, 2025-01-04:
@ajgowans @sorgelig  this issue can be closed. It appears to have been implemented correctly via https://github.com/MiSTer-devel/SNES_MiSTer/commit/5536103b652a927dead813e44b7cec9aac376267

Testing via https://github.com/MiSTer-devel/SNES_MiSTer/issues/393

closed 2025-01-06 by sorgelig