MiSTer-devel/SNES_MiSTer #281: Barts nightmare runs faster than snes hardware

pile203, 2021-05-17:
watching someone do speedruns of barts on a mister i noticed that the mister handles lag better than the snes causing it to be faster gameplay. after some brief play testing, i have created a side by side comparison video to show the difference in snes hardware and mister starting on the same frame of screen scrolling to the next area.
https://youtu.be/njVxk75kG_U

birdybro, 2021-06-11:
Can you compare with the latest nightly?

https://github.com/MiSTer-unstable-nightlies/SNES_MiSTer/releases/download/unstable-builds/SNES_unstable_20210507_fca7.rbf

It might be related to some recent fixes since the release in February.

sorgelig, 2021-06-11:
i think it's the same issue as: https://github.com/MiSTer-devel/SNES_MiSTer/issues/208

birdybro, 2022-05-22:
Your side by side isn't synced up to the frame exactly. Since issue #208 has been significantly improved upon, can you redo your test and sync to the exact same starting frame and compare again please?

However, the real SNES hardware differences have a very wide frametime variation that is unavoidable, so multiple pieces of hardware should be compared as well.