MiSTer-devel/SNES_MiSTer #286: Black Screen for "Itoi Shigesato no Bass Tsuri No. 1"

mr-kyubit, 2021-07-01:
Hey there! I'm having an issue loading the game "Itoi Shigesato no Bass Tsuri No. 1 (Japan)". The ROM seems to load (I get the standard progress bar), but after that I just get a black screen. The SNES core compatibility has been incredible, and so far this is the only game I've had trouble with. I'm using the version included in the No-Intro set, and the ROM works fine on bsnes. 

One interesting note: the original game came as a slotted cartridge with an optional memory pack for Satellaview transfers (https://satellaview.fandom.com/wiki/Slotted_cartridge), but so far I've been able to run other slotted cartridge games just fine in the core (ex: "SD Gundam G-Next" and "Sound-Novel Tsukūru"). I realize memory packs are not currently supported in the core, but in principle the game should load without a memory pack. For example, when I open the ROM in bsnes, there's a memory pack option on the main menu of the game, but it's greyed out.

I posted this to the core-bugs channel on the Discord, and wark91 pointed me to a similar ticket they opened back in October regarding two other slotted cartridge games that appeared to have the same issue: #245 

Thanks in advance for taking a look at this!

MP2E, 2024-10-09:
I’m not sure when this was fixed, but I tested it out on the latest SNES unstable core and everything appears to be working!

paulb-nl, 2025-06-29:
Can be closed.

closed 2025-06-29 by sorgelig