MiSTer-devel/SNES_MiSTer #288: Feature Request: Save state support

mlc4050, 2021-07-09:


SegaSnatcher, 2022-12-09:
> Is this feature planned at all? It has some implications for other platforms, such as the Analogue Pocket, which uses save states for its suspend/sleep support. See [agg23/openfpga-SNES#59](https://github.com/agg23/openfpga-SNES/issues/59)

The core was not initially developed with save states in mind, so it would take a lot of work to add them.   Basically, the main thing that needs to happen is the ability to pause the core which in of itself would require a ton of rework of the core itself before save states could even be added.   Robert is likely the only person with the skills to take up such a challenge, but he's really busy with the PSX core and likely will be for the foreseeable future. 

So, I would not expect save states to be added anytime soon, unless someone with the skills and a love for a challenge shows up.

littlegamer87, 2023-01-14:
I made an enquiry to Robert about this, he has declined as he knows nothing about the core. 

https://twitter.com/iamTeddyNL/status/1614234810536398848

birdybro, 2023-10-31:
@sromeroi - Just for your awareness, every developer knows lots of people wants savestates in these popular cores. Every reply and every new issue on every single one of the MiSTer repos ends up spamming sorgelig's email inbox, so please be considerate.

cheskote, 2023-10-31:
+1 for this feature request

paulb-nl, 2024-10-08:
I posted a core with save state support on the forum so people can test it if they want.

https://misterfpga.org/viewtopic.php?t=8459