MiSTer-devel/SNES_MiSTer #290: Blue tinted background entering second race (and after) in Top Gear 3000

lhathcock, 2021-07-21:
Hopefully this isn't a repeat of another issue, but I've run into an issue where the map screen has a blue tint to it when you complete the first race and view the second map screen. It also sticks around for courses afterwards as well. Apparently this was also an issue in the Super Nt, and I'm not sure if it was fixed there or not.

Relevant link: https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/74

MP2E, 2024-10-02:
I just tested this with the latest changes as of 20241001, and the map screen for the 2nd race seems to be the correct color now!

Not sure if I just got lucky, could anyone else confirm?

paulb-nl, 2025-01-27:
Played a few courses with build 20250119 PAL and NTSC and did not get the blue background.

closed 2025-01-28 by sorgelig