MiSTer-devel/SNES_MiSTer #296: Super Metroid Practice Hack

radon0303, 2021-08-29:
The Super Metroid practice hack with save states does not currently work on the SNES core. The save states are unique to this rom and are implemented as follows: all of the WRAM, VRAM, and CGRAM are saved to the SRAM so you need a large enough SRAM to allow for this to happen. The SD2SNES has expanded SRAM and so it allows for the implemented save states to work. This feature currently works on the Super NT as well. I was told by Kitrinx that the SRAM allocation in the SNES core isn't currently large enough for this to work because a lot of the BRAM is used for the special chips. I know the fix wouldn't be trivial, but adding support for the hack would allow a lot of speedrunners to move over to the MiSTeR so it would be greatly appreciated.

Link to the rom is [here](https://smpractice.speedga.me/), link to the git page for the rom is [here](https://github.com/tewtal/sm_practice_hack/), and [here](https://github.com/tewtal/sm_practice_hack/blob/master/src/save.asm) is the link to the code where the save states are implemented.

neovol, 2021-09-09:
I have the issue as well and can vouch that it would be great to have this available to us MiSTer SM players. Please let me know if there is anything I can provide to help troubleshoot this.

charlesmadere, 2021-09-13:
Being a Super Metroid speedrunner and streamer myself, I've recommended the MiSTer many times to viewers who ask about ways to get into the game and don't want to use/don't enjoy using an emulator. However, I always make sure to make it clear that the SM practice hack itself currently does not work on the MiSTer, as it could be a dealbreaker for these new runners. (it would definitely be a deal breaker for me)

Fingers crossed 🤞 that we hear about some updates on this. Thank you!!

dannydrogt, 2021-09-13:
Going to keep this comment short as it pretty much is exactly the same as @charlesmadere already mentioned, but I actually do own a MiSTer and would love to see the practice ROM work on it with the save states!

I'm pretty sure a LOT of runners from the community will switch to a MiSTer when this works, so I hope you can get us there!

radon0303, 2021-09-22:
I spoke with one of the practice hack devs and was told that an SRAM size of 256K would be enough for the hack to function properly. Also had a short conversation with kevtris about it to see if I could provide any more info that would be helpful to the issue. He said that there were 64 banks made available for SRAM, 0-3F and 6000-7FFF. So this left a total possible amount of 512K-16K=496K of space. This is mirrored at banks 80-BF, which is another 64 banks, so it can be expanded further. I'm not sure how much space is made available for SRAM on the Fxpak. Will do some research on that as well and post what I can find.

radon0303, 2021-10-11:
Just updating with more info: Spoke with Ikari and he said that the theoretical limit should be 4Mbits for the SRAM and on the SD2SNES the currently mappable amount if 1Mbit.

birdybro, 2022-04-23:
@charlesmadere You mentioned that the MiSTer SNES core is "out of spec". What do you mean specifically? It would be helpful to have some of the info provided here so it could be remedied, if there is a problem. The Super Metroid Practice Hack seems to take advantage of the theoretical banking potential of the SNES hardware that was never utilized contemporaneously. The SRAM space could be expanded in the core, but probably not to 1Mbit. How much space does the hack require?

sorgelig, 2022-04-23:
almost all BRAM (SRAM) is already used in SNES core. If hack requires more then it's impossible to make. It doesn't make SNES out of spec. Core is made to work with normal games. Hacks may implement cartridges never used on SNES in its lifespan - it's out of spec. You may even implement cart with Core i5 inside - it doesn't mean it should automatically work with this core. So before blame the developers, think more.

charlesmadere, 2022-04-23:
@birdybro @sorgelig unfortunately I myself am not personally extremely knowledgeable on the intricacies of the Super Metroid practice rom's SRAM requirements other than that the developers of the practice rom themselves have told me that the hardware requirements it asks for are wholly within the SNES spec. Additionally, the Super Metroid practice rom does work totally fine on the [Super Nt jailbreak firmware](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak).

But regardless, in the meantime, it's the users of the Mister who lose.

charlesmadere, 2022-04-23:
Above, @radon0303 left some technically information that looks like it could be a good start: [[one](https://github.com/MiSTer-devel/SNES_MiSTer/issues/296#issuecomment-925302222)], [[two](https://github.com/MiSTer-devel/SNES_MiSTer/issues/296#issuecomment-940050742)]

paulb-nl, 2022-04-23:
It is possible to move Cartridge RAM to the SDRAM for games without special chips so it can be larger. 

Games that have special chips like Super FX can usually read from Cart RAM while the CPU reads from ROM so those need BRAM but for games like Super Metroid it should be possible to use the SDRAM for Save RAM.

However this is not easy to do for this core.

The NES core already uses SDRAM for Cart RAM.

dannydrogt, 2022-04-23:
I know absolutely nothing of hardware/specs/etc, but I do want to quickly add to this topic that the .SRM file I see on the SD2SNES for the Super Metroid practice hack is indeed 256k.

The comment above from paul is what I've read about on the MiSTer Discord as well (I have talk there in the past with Kevtris about this issue). It's interesting to know the NES core already does do what we'd actually want the SNES core to do for getting the practice hack to work?

birdybro, 2022-04-23:
As paulb-nl said, it's not easy to do for this core. It would likely entail changing the role of the BRAM reserved in the current core to "catch" when a game is using it in the way Super FX games do, and all else uses the SDRAM method. Or creating a mirror of what is in BRAM in SDRAM and branching the path behaviorally, but I don't believe any of this is trivial to do in FPGA designs. Possibly quirks could be added, but they are undesirable because it's a slippery slope to having 100 quirks one day.

radon0303, 2022-04-27:
@birdybro when I spoke about this issue in discord the "catch" method was brought up as a solution as well. It's interesting to know that this is already implemented in the NES core. What specifically about the logic used in the case of the Super FX chip makes it hard to extend to each of the other special chips? @sorgelig by out of spec we just meant that allocation was different than what Nintendo allowed when things were designed, didn't mean it as a slight towards the core. I'm thankful for the work everyone has put in.

birdybro, 2022-04-27:
What specifically? I don't know the specifics or else I'd try to do it or something :P 

https://github.com/MiSTer-devel/SNES_MiSTer/blob/master/rtl/chip/GSU/GSU.vhd

https://github.com/MiSTer-devel/SNES_MiSTer/blob/master/rtl/chip/GSU/GSU_PKG.vhd

The specifics of how the Super FX chip works in this core now, are mostly in there. It doesn't look easy to read through this and then understand the implications of modifying the code completely. Remember this programming language describes digital logic, tiny circuits. It's not as simple as just adding more if/else statements. The result can negatively impact timing, they can take up so many resources that necessary tools like SignalTap become impossible to use, etc... So generally speaking, Paulb-nl knows what he's talking about when he says it's hard, so I trust his judgement.

pgambardello, 2022-05-03:
I would like to say I would also love to see the SM practice ROM working.  An alternative to a Super NT/FX Pack combo that is open source would be much appreciated.  TY

hypnogram, 2022-05-04:
Commenting to say that I would ALSO love to have this.

radon0303, 2022-05-04:
@birdybro This makes a lot of sense, thanks for elaborating. When I asked for specifics didn't mean to imply that it would be trivial, was just interested in the technicalities because I'm not familiar with HDLs.

Drakonas, 2022-05-16:
> @birdybro This makes a lot of sense, thanks for elaborating. When I asked for specifics didn't mean to imply that it would be trivial, was just interested in the technicalities because I'm not familiar with HDLs.

Basically the SNES core is so full because it has a couple handfuls of special chips, some as advanced as the SNES CPU itself, that it has to have coded support into the core for the FPGA. There's limited space in an FPGA, so there are limits.

I think the confusion of "out of spec" is that what's possible on the hardware doesn't define it being in spec or not. "Out of spec" implies what was released at the time of the console's lifespan. Having a game that would do such a thing in the 90's wasn't cost effective. And even if it was, not every developer would even know how to pull the bank-switching technique off on that level. So it's not really accurate for a clone of the hardware to be able to run something like it did in the 90's when the rom wasn't technically plausible to be released in the 90's.

The other issue is the SD2SNES had a PCB designed for the SNES specifically, so I believe it is able to implement more SRAM by design using an actual chip. I believe it has a 32MB (2x16MB Parallel) PSRAM chip in its design, that is connected directly for use by the FPGA. And the FPGA itself can have internal RAM blocks as well.

Meanwhile, the MiSTer is running its own MiSTer core in the background while the main console core is running.

I don't want to undermine any efforts by the team. Praise them if they're able to do this, but I wanted to explain that there are a lot of factors for why this is a lot harder to pull off on the MiSTer than the SD2SNES. The SD2SNES was made forward-thinking for SNES, so it has a bit of an advantage for SNES over a De10-Nano that wasn't even made with the MiSTer in mind. It's just the board that was chosen for the project.

I just wanted to shed some light on how there's more to these boards than just raw numbers or skill. FPGA's simulate hardware electrically in a single chip, instead of emulate them with pure software processing. The regular SD2SNES has about 75,000 logic gates it can provide entirely to the SNES, whereas the De10-Nano has 110,000 for the entire MiSTer, main core and the running console/computer/arcade core. And because the De10-Nano doesn't have the advantage of having custom CIC chips and such on a PCB like the SD2SNES, they have to be implemented in the FPGA as well. The SD2SNES has a lot of separate chips handling other parts, to allow more room for intricate stuff in its FPGA.

I hope I didn't bore anyone. I think FPGA's are fascinating pieces of tech, even though programming is not my strong suit.

birdybro, 2022-05-16:
Yup, this is a good summary. BlockRAM inside an FPGA is a lot more costly and sparse when compared to SDRAM/PSRAM/HyperRAM on these flash carts. It's not as much a question of space in the blockRAM, it's the timings of running the game from SDRAM while having a shadow copy of the game's SRAM inside the SDRAM at the same time. (if I understood paul correctly)

tewtal, 2022-05-20:
Specifically regarding this issue I've started working on a change to the Super Metroid Practice hack instead to work within the limitations of 128k. This doesn't help any other case where 256k SRAM is required, but it'll help for those wanting to use the save state features of the SM practice hack.

I've made a draft PR on the repo here: https://github.com/tewtal/sm_practice_hack/pull/94
There's an IPS patch attached to the PR for testing, please report any issues found with it there.

threeandthreee, 2023-10-30:
Out of curiosity, would this be more doable on a Super Metroid specific fork of the core that drops support for other special chips to save space?

kandowontu, 2024-08-19:
As littlegamer87 mentioned, 256KB is confirmed working on hardware. Hoping with more hacks/homebrews/ports requiring this, would it be possible to support it?

birdybro, 2024-08-19:
It's possible, but not easy to do, read paulb-nl's comment above. +1'ing this issue is kind of pointless, it's still open and could be addressed at some point if a dev wanted to do it.

kandowontu, 2024-08-20:
> It's possible, but not easy to do, read paulb-nl's comment above. +1'ing this issue is kind of pointless, it's still open and could be addressed at some point if a dev wanted to do it.
> 
> EDIT: Just doubled the sram and because of the way it's used there are not enough M10k blocks for it to successfully compile, so it's just like paulb-nl said.

So basically a major rewrite. Got it.

Drakonas, 2024-08-21:
> So basically a major rewrite. Got it.

Yep. Wanted to add a little extra to my big block of text above.

People often ask why things can fit in the smaller FPGA of the SD2SNES with 75k logic blocks and not the MiSTer with 110k logic blocks.

This has a simple answer. The SD2SNES is not housing any of the SNES hardware in its FPGA. It only has to deal with the cartridge-level hardware. So this means any of the special chips, and IO for the various ports on the flashcart.

Thus, the MiSTer core has to be far larger to house both the main SNES hardware and the special chips (which it supports more of compared to SD2SNES I might add).

Just wanted to clarify that. But yes, this isn't "impossible" from what we can tell, but it would take a lot of effort and thus would need a massive amount of bugtesting and new compatibility issues with games that would have to be reported and fixed.

This happened with the MegaDrive, so perhaps some day someone might. Having the ability to connect the MiSTer to the Super Metroid and Link to the Past randomizer item tracking PC software would be really neat if someone could develop a serial connection to use a similar debug port for it like the SD2SNES has.

Maybe some day, but yeah there's no reason to resurface this issue. It's noted. People are generally aware. It's just a lot to ask. Haha.

jdharms, 2024-09-23:
I think this can be resolved now that 74d7761d549f5bb81ef15d30bf5e136edbc0ebaa is in?

MP2E, 2024-09-23:
Yes I tested the super Metroid practice hack and save + load savestate worked perfectly.

closed 2024-09-24 by sorgelig