MiSTer-devel/SNES_MiSTer #298: Super Mario World intro chime/screen & Link To The Past intro chime/screen errors

BretYancy, 2021-09-05:
i have noticed, for quite a while, the intro chime in both super mario world and link to the past sometimes doesnt play. 
or, the chime will play, but the intro image/screen is still blank on my screen. if i reset the game after the game is loaded, the intro image/screen will show _sometimes_, but never the first time the game is loaded from the mister menu.

im running my mister over HDMI, low lag mode, integer scale .25, with everything up to date.
(i have not tested on a CRT, but the analogue super NT does not have this issue over HDMI in low lag mode)

my thought is that there is a sync/refresh switch happening when the game is first loaded, or something is re-syncing/refreshing once a game is loaded.

**is there any way to correct this?**
perhaps having an option to delay the start of the ROM by a couple seconds, giving the screen time to sync/refresh??

BretYancy, 2021-09-09:
**update**, 
this is happening on both the screens i tested, but others have reported this is not happening on their older LCD monitors. so at this point, i feel like this is more of a feature request for the SNES core

perhaps having an option in the core to delay the start of the ROM by a couple seconds, giving the screen time to sync/refresh?
can someone try compiling a version of the core that implements a lock on the refresh rate or the above option to start the ROM a few seconds later to give the screen time to "catch up"?
like, make an option in the core to load a game/video signal thats 60.08hz or whatever the SNES refresh rates is for a couple seconds, which will allow monitors time to sync to the core?

to clarify, this is the screen that gets missed if i have my MiSTer set to vsync_adjust=2 over HDMI -
https://www.youtube.com/watch?v=wejcLMtvntw

and again, this screen **does** appear if i set vsync_adjust=1

SegaSnatcher, 2021-11-06:
This is a pretty common issue using v_sync=2.   Most displays will simply not sync right away and so you won't hear the chime or see video right away.  Its not really a MiSTer issue, but simply how displays handle non standard refresh rates.  Pretty sure this issue can be seen when using an OSSC on some displays as well.

bikerspade, 2024-09-25:
@BretYancy this does not seem like it is an issue specific to the SNES core, but rather related to how your displays handle various refresh rates. I think this issue can be closed.

closed 2024-09-26 by sorgelig