MiSTer-devel/SNES_MiSTer #300: GRS-USB / Atari USB spinner partially incompatible w/ SNES mouse support

MukiEX, 2021-09-21:
I have a GRS-USB, which seems to have a similar firmware to the Atari USB buttonhole spinner device. (a MisterFPGA forum user had the same issue) It can be wired to two additional pushbuttons which function as the left and right mouse clicks.

In both cases, they both show up as a "Mr.Spinner" paddle device. On a PC they show up as a USB mouse. As they're spinners, they can only move in one axis at a time. Holding the left/right mouse buttons for 10s will swap that axis.

MiSTer's Arcade Arkanoid, Pong, and NES cores recognize and function correctly with the device. On the SNES and Genesis (just checked) cores, the mouse buttons work, and the MiSTer dialog for a Mr.Spinner pops up, but the left/right movement for the paddle does not work.

sorgelig, 2021-09-21:
SNES doesn't support paddles at all.
Also, you cannot use this spinner as a mouse because MiSTer recognizes both these spinners and allow them work only as spinners.

MukiEX, 2021-09-22:
For what it's worth, and just because I'd like to make my case:

(extra note: I'm 100% prepared to go figure this out how to implement this myself, rather than bug you about it, and will update this issue with a potential merge request provided my code doesn't turn out to look like an absolute mess, but I'm a pretty low-tier coder. I haven't messed with the MiSTer framework much and I barely know my way around an Arudino, so it could be some ways away)

Spinners and Driving Paddles (e.g. paddles that don't stop in one direction or another) have basically been used as replacements for one another as consoles progressed over the years, mostly because potentiometer-based paddles have had much smaller availability for officially-supported controls than mice were. Probably the golden example (and basically the game I'm kinda brushing up against here) being the various ports and sequels to Taito's Akranoid.

The arcade machine for Arkanoid used a multigear rotary encoder, which, for all intents and purposes, functioned nearly identically to how early ball mice did, internally. A rectangle-spoked gear would spin past an optical sensor, just like they did on ball mice to determine left/right movement.

The NES port of Akranoid could use the Vaus controller, which AFAIK was driven by a potentiometer on the knob, much like the early Atari paddle controllers.

The SNES iteration of Akranoid, e.g. Doh It Again, in leui of a paddle controller, mapped left/right movement to the X axis of an SNES mouse, retaining the analog functionality of the arcade-style movement.

The Playstation sequel also used the Playstation Mouse, as did Tempest 2000 (a sequel to the arcade spinner-based Tempest game), along with Puchi Carat (another Taito game that used the same spinner knobs as Arkanoid)

On MiSTer, a Mr. Spinner device will work on a the NES and Arcade cores, but not SNES. A USB mouse will work on the Arcade and SNES cores, but not the NES. It's a little frustrating as while the implementation is different (e.g. a paddle will send absolute values across an X axis for its potentiometer, while a mouse and most spinners will send horizontal X/Y movement), the intended behavior is nearly identical. (Turn knob counter-clockwise, move paddle left. Turn knob clock clockwise, move paddle right)

While I did get a single device working across all three, it basically relied on using Arduino's device reporting feature to basically lie to the MiSTer unit and report itself as either a mouse or Mr. Spinner depending on the mode is was started up in. Aside from that odd workaround, there's no single controller that will work as intended for 3 different ports/sequels of a game that were basically designed to control identically.

All I really had to do was "internally" turn the potentiometer value with the X/Y movement, and was even able to add a sensitivity controller care of a potentiometer wired to the arduino board, and the MiSTer cores already have very similar functionality for mapping analog stick positions to paddle/spinner controls in some cores.