MiSTer-devel/SNES_MiSTer #303: Top Gear - graphical glitches with sprite occlusion on the road

Kopert, 2021-11-14:
I don't know if sprite occlusion is the correct term, I don't know how it is actually implemented. Anyway:

**Core**: SNES_20210713.rbf
**ROM**: NO-Intro, Top Gear (USA).sfc, MD5 e5040a079ff03eb93747424af9be75f4

**Issue**: Sometimes sprites still show up when they're supposed to be occluded by the road, for example, when going downhill.

Here are two images that show the problem:

[image]
[image]
 On these images, taken near the last turn on the first level (Las Vegas, U.S.A), you can see a left-turn signal that should be hidden by the road. I've noticed this on other levels with other sprites, including cars.

I do not have an actual SNES to test this one, but I have checked [this video](https://www.youtube.com/watch?v=kCeRTjxli88) taken from the actual console and I've tested it on the bsnes_mercury_accuracy RetroArch core, and the issue does not appear on either case.

Kopert, 2021-11-21:
I don't know if this helps, but I tested `SNES_20210713.rbf`, `SNES_20210620.rbf`, `SNES_20210225.rbf`, and `SNES_20201106.rbf` (just a random sampling) and I could reproduce the issue in all of these.

paulb-nl, 2021-11-22:
I was able to reproduce the issue and I can confirm that it doesn't happen on a real SNES.

I have been looking into how the game hides parts of the sprite but I don't understand it yet.

Kopert, 2021-11-22:
> I was able to reproduce the issue and I can confirm that it doesn't happen on a real SNES.
> 
> I have been looking into how the game hides parts of the sprite but I don't understand it yet.

Thank you for taking your time and confirming the issue. Let me know if there is any way I can help.

paulb-nl, 2021-11-23:
The game hides parts of sprites by adding many off-screen sprites that still take up time to fetch so that there is no time left to fetch the sprites which should be hidden.
[image]

These off-screen sprites are 16x16 with X position -256(or 256) so these should take 4 pixel cycles to fetch but the core only takes 2 pixel cycles to fetch these sprites.

This trace shows sprite 46 takes 4 pixel cycles to fetch which is correct because it is a 16x16 sprite with X position 97. Sprite 47-51 are 16x16 off-screen sprites but only take 2 pixel cycles to fetch which should be 4 cycles.
[image]

Here is a save state for Mesen-S:
[Top Gear (U)_2.mss.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/7589243/Top.Gear.U._2.mss.zip)

sroccaserra, 2021-11-23:
Nice, thank you for this info!

paulb-nl, 2021-11-24:
For reference, here is a capture of a SNES. It looks the same as with the save state of Mesen-S. The car to the left of the white car is what causes the missing sprite lines in the white car.
[image]

paulb-nl, 2021-11-26:
There is a check here that skips to the next sprite if the next tile is off-screen. The `or (TILE_X + 8) >= 256` part.
https://github.com/MiSTer-devel/SNES_MiSTer/blob/be0cfc883443d9bf714243edfcffd1cba40cee11/rtl/PPU.vhd#L1569
@srg320 Do you remember why this check is done? It doesn't seem to match a real SNES with Top Gear.

If I remove `or (TILE_X + 8) >= 256` then the result is like a real SNES:
[image]

@Kopert Can you test this build with other games?
[SNES_sprite.rbf.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/7608846/SNES_sprite.rbf.zip)

Kopert, 2021-11-26:
> @Kopert Can you test this build with other games? [SNES_sprite.rbf.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/7608846/SNES_sprite.rbf.zip)

Thank you for the build, @paulb-nl. I tested it on about a dozen games keeping my eyes up anything that looked out of place and couldn't find anything.

<details>
  <summary>Click to see list of games tested on 2021/11/26 with SNES_Sprite.rbf</summary>

- Top Gear - fixes https://github.com/MiSTer-devel/SNES_MiSTer/issues/303
- Doom - ok (still affected by https://github.com/MiSTer-devel/SNES_MiSTer/issues/295)
- Cannondale Cup - ok
- Super Mario World - ok
- Super Mario World 2: Yoshi's Safari - ok
- Yoshi's Safari - ok
- Wario's Woods - ok
- Rock n' Roll Racing - ok
- Flashback - ok
- Jurassic Park - ok
- A.S.P.: Air Strike Patrol - ok
- Mario Kart - ok
</details>

My main concern is that the issue with Top Gear was very hard to perceive in the first place, so there's a chance that any issues your changes would introduce would easily escape my notice. Ideally we'd know why the `or (TILE_X + 8) >= 256` was added in the first place. It might have been just some optimization that does not reflect real hardware, but it might have been added to fix some other issue.

I will continue using your build as my daily driver so if anything out of place pops up I'll let you know. Thanks again!

paulb-nl, 2021-12-02:
The sprite test rom I made does show that the SNES only fetches the visible tiles so that's why that TILE_X+8 check is there. 

Only when a sprite is set to X position -256/256 then it does fetch all the tiles even though the sprite is completely off-screen.

Kopert, 2021-12-02:
> The sprite test rom I made does show that the SNES only fetches the visible tiles so that's why that TILE_X+8 check is there.
> 
> Only when a sprite is set to X position -256/256 then it does fetch all the tiles even though the sprite is completely off-screen.

I'm out of my depth here, but would that be a case of replacing `(TILE_X + 8) >= 256` with `(TILE_X + 8) > 256` then?

paulb-nl, 2021-12-04:
That doesn't work because `TILE_X` equals 256 on the first tile of a sprite with X position 256.  256+8=264.

Also it would not skip the next tile of a sprite if the next tile starts at 256.

Kopert, 2021-12-10:
I just tested SNES_unstable_20211210_c8c9, which includes 3c622c1 (fixes #306), and I can confirm the Top Gear issue is fixed on this build. Thanks, everyone, for your time!

Should I close the issue, or is there need for additional testing or development?

sorgelig, 2021-12-10:
better to wait for release, then test and close if fixed.

Kopert, 2021-12-14:
Tested with release 21.12.14. Could not reproduce the issue anymore. Thanks for the fix!

closed 2021-12-14 by Kopert